robotology/yarp #470: yarpdataplayer (Qt) does not work with the withExtraTimeCol option

matejhof, 2015-04-03:
I was replaying datasets that use the yarpdatadumper --txTime --rxTime option (two time stamps). The yarpdataplayer with either --withExtraTimeCol 1 or --withExtraTimeCol 2 does not work. In both cases puts the extra time stamp on the port as data.
The same datasets replay fine on the yarpdataplayer-gtk. 
Maybe this feature was not implemented correctly during the porting to the Qt version?
@vtikha can help perhaps?

vtikha, 2015-04-03:
Hi @matejhof , are you up to date with the YARP repo? I have just tried to replicate this with some data set that uses txTime and rxTime (that you once gave me actually) and --withExtraTimeCol 1 or 2 works as it should. Could you please check if you are up to date with YARP, that you are not using some installed version of the yarpdataplayer (do "which yarpdataplayer") and send me a link to the dataset you are trying to replay for me to try?

matejhof, 2015-04-03:
Sorry, false alarm. I was using '-' instead of '--' the option in the yarpmanager window. Unfortunately, they look very very similar there. So it was calling it without it. Sorry again, closing.

closed 2015-04-03 by matejhof

vtikha, 2015-04-03:
-1 for @matejhof ;-)