robotology/yarp #473: getPendingReads() & flush ?

stephane-lallee, 2015-04-13:
Hi !

I recently came upon a behavior not expected. Like if the buffered port was accumulating some data that would then be consumed by the consecutive reads.

I feel that this behavior is somehow new, am I right ?

Anyway, I digged a bit and found this new (to me) method getPendingReads(), that I then used to code a flush to drop all the messages before a read that I wanted to be clean.

My questions are : 
- is there already a flushing mechanism somewhere else ?
- do we have any way to parametrize the length of the buffering window or the number of pending reads that can be hanging ?

Thanks in advance !
Stephane

pattacini, 2015-04-13:
Hi @stephane-lallee 
Well, the original behavior of a `BufferedPort` hasn't changed, obviously: could you imagine the effects on all our software?

As quick test I've written two code snippets in `lua`: a transmitter running @ 10Hz and a receiver polling at the lower rate of 1Hz

##### The transmitter:
```lua
#!/usr/bin/lua

require("yarp")
yarp.Network()

port=yarp.BufferedPortBottle()
port:open("/tx")

N=500
for i=1,N do
    local wb=port:prepare()
    wb:clear()
    wb:addString("count")
    wb:addInt(i)
    wb:addString("of")
    wb:addInt(N)
    port:write()
    print(wb:toString())
    yarp.Time_delay(0.1)
end

port:close()
yarp.Network_fini()
```

##### The receiver:
```lua
#!/usr/bin/lua

require("yarp")
yarp.Network()

port=yarp.BufferedPortBottle()
port:open("/rx")
yarp.NetworkBase_connect("/tx","/rx")

N=3
for i=1,N do
    local b=port:read(false)
    if b~=nil then
        print("received: " .. b:toString())
        print("pending: " .. port:getPendingReads())
    else
        print("nothing received")
    end
    yarp.Time_delay(1)
end

port:close()
yarp.Network_fini()
```

The outcome is as expected: no flush is required and the freshest data are always returned by the receiver.

stephane-lallee, 2015-04-14:
Ha... I see what I'm doing. There was actually one last message pending and I was reading in a loop waiting for user speech, so it was getting the last message indeed.
My bad, sorry. At least I discovered this new method, can be usefull indeed ! 😊
I'll close the issue, thanks !

-----Message d'origine-----

closed 2015-04-14 by stephane-lallee

pattacini, 2015-04-14:
Great