robotology/yarp #474: yarpview and yarpview-gtk are inconsistent

apaikan, 2015-04-15:
yarpview-gtk and yarpview (Qt) are not consistent?   

**yarpview --name /view**
**yarpview-gtk /view**

- Setting the port name for `yarpview-gtk` 
```
$ yarpview-gtk /view
testing name: /view
yarp: Port /view active at tcp://192.168.1.158:10002
```

- Setting the port name for `yarpview` 
```
$ yarpview /view
yarp: Port /yarpview/img:i active at tcp://192.168.1.158:10003
```

alecive, 2015-04-15:
Yes they are not consistent.
Quoting @pattacini from https://github.com/robotology/yarp/issues/328 : "with the new qt version (and only with the qt version), it's been decided to harmonize the parameters passing."

alecive, 2015-04-15:
You can also take a look at https://github.com/robotology/yarp/pull/242  for this.

closed 2015-04-15 by alecive