robotology/yarp #475: writing to the port under PortReport::report() callback

pattacini, 2015-04-15:
The following code snippet

```cpp
#include <yarp/os/all.h>
using namespace yarp::os;

class ReceiverPort : public BufferedPort<Bottle>
{
    void onRead(Bottle &b)
    {
        yInfo()<<"received: "<<b.toString().c_str();
    }
} rx;

BufferedPort<Bottle> tx;
void writePort()
{
    Bottle b("test");
    yInfo()<<"sending: "<<b.toString().c_str();
    tx.prepare()=b;
    tx.write(true);
}

class TestReporter : public PortReport
{
    void report(const PortInfo &info)
    {
        if (info.created && !info.incoming)
        {
            writePort();
        }
    }
} testReporter;

int main()
{
    Network yarp;

    rx.open("/rx");
    rx.useCallback();
    
    tx.open("/tx");
    tx.setReporter(testReporter);    
    
    Network::connect(tx.getName().c_str(),rx.getName().c_str());

    //writePort();
    Time::delay(0.1);
    
    rx.close();
    tx.close();

    return 0;
}
```

does attach a `PortReporter` object to a transmitting port so that upon connection a bottle is written to the same port and we can log the message received.

Well, this code does not work, getting stuck when `write()` is called. Conversely, if we comment out the first call to `writePort()` and then enable the second call, everything runs smoothly.

Is this an expected behavior? Might this be happening because when the callback `report()` is executed the connection is not properly in place yet?

/cc @paulfitz @lornat75 @drdanz @alecive

paulfitz, 2015-04-16:
If I'm reading the code right, this may be expected.  From `PortReport::report`:

> This callback must be careful about performing any operation on the port it is receiving information about, since that port is locked until this callback completes. To avoid deadlocks, take action based on port reports in a separate thread.

A write operation could result in a bad connection being discovered and dropped, resulting in the callback.  There would be ways around this, but it is awkward...

pattacini, 2015-04-16:
Cool, thanks @paulfitz 
Actually I didn't realize that what I was looking for was already written in the documentation.

closed 2015-04-16 by pattacini