robotology/yarp #476: yarpMotorGui - Calib_all button

barbalberto, 2015-04-16:
"Calibrate all" button should be disabled by default, unless program is started with --calib option.
It works for calib button on single joint, the same mechanism should be implemented also for upper button.
[image]

btw, I cannot see the icon for bottons located near that calibrate all button :'(

pattacini, 2015-04-16:
The appearance of buttons might be related to #460

aitek4iit, 2015-04-16:
Fixed The enable problem on the button

alecive, 2015-04-17:
The appearance of icons is related to a missing library. You need to install `libqt5svg5` in order to see the new icons (that are in svg format). I updated the documentation some weeks ago, you can see it here: http://wiki.icub.org/yarpdoc/dependencies.html

lornat75, 2015-04-17:
@mbrunettini

barbalberto, 2015-04-17:
Great, both the issues are fixed,

thanks @aitek4iit  and @alecive

closed 2015-04-17 by barbalberto