robotology/yarp #478: Can't send message from ROS to YARP

mikearagao, 2015-04-20:
I'm getting the following message while trying to read the message from YARP:

    WireTwidderReader shrugging

The line of code that's generating the following error is this one:

    SimpleTrajectory* msg = port.read();

Any help?

mikearagao, 2015-04-20:
UPDATE:  that's not the line causing this error... trying to debug it now.

barbalberto, 2015-04-21:
Hi @mikearagao,

can you post a full example of the code you are using (just relevant code for the communication) and which ROS msg are you trying to read?
If I can replicate your case it'll be easier to debug.

mikearagao, 2015-04-24:
Hey @barbalberto! Sorry I haven't been really busy because my thesis deadline is near... I'm trying to read a simple message that I've created, you can check it here: https://github.com/vislab-tecnico-lisboa/yarp-with-moveit/blob/master/ros_code/miguel_vislab/msg/SimpleTrajectory.msg

The YARP code is this one: https://github.com/vislab-tecnico-lisboa/yarp-with-moveit/blob/master/yarp_code/apply_trajectory.cpp

Cheers!

barbalberto, 2015-06-26:
Hi @mikearagao, sorry for not replying for such a long time. Do you still have problem with this issue?

mikearagao, 2015-07-19:
Hey @barbalberto sorry for the late reply. @gsaponaro sent me an email asking if I was able to solve this and I just replied to him.

It happens that I have left the lab (now working on a startup doing web dev) but I have  been trying to help on some unfinished work and this specific issue is solved! I was trying to run code that was not compatible with the most recent versions of YARP.

I hope everything is fine with the team!

Cheers

barbalberto, 2015-07-20:
Ok, closing this issue.
More info in yarp-ros interoperability in the issue #538 
If you still have problems take a look there and feel free to reopen the issue if it does not help.

closed 2015-07-20 by barbalberto