robotology/yarp #481: YARP Bindings do not get installed

Tobias-Fischer, 2015-04-21:
Hi all,

I found a little bug: Bindings of YARP (I only tested LUA) do not get copied to the installation folder when executing ```make install```.

The way this bug can be replicated is as follows:
1. Switch ON the following flags: YARP_COMPILE_BINDINGS; CREATE_LUA
2. ```make``` -> bindings are compiled in build/lib/lua
3. ```make install``` -> bindings should be copied to CMAKE_INSTALL_PREFIX/lib/lua, but are not

Would be great if this could be fixed.

Thanks, Tobias

drdanz, 2015-05-12:
It looks like that python and allegro bindings are installed, all the others are not

lornat75, 2015-05-21:
However it still requires to set LUA_CPATH, i.e.:

LUA_CPATH=;;;<INSTALL_PREFIX>/bin/?.so

drdanz, 2015-05-21:
From what I understand it should not be installed in "bin" but in "lib/lua/5.2" or something similar, and I think that you will not have to export LUA_CPATH...

drdanz, 2015-05-21:
I just noticed that bindings install examples and other stuff that is not supposed to be installed, but do not install the actual target... we should definitely clean up this:

* [ ] Move the bindings examples with the other examples
* [ ] Disable out of source compilation, simplify CMake, remove YARP_COLLISION_AVOIDANCE, etc.
* [ ] Install in the right place all the bindings.

lornat75, 2015-05-22:
Problem with installation in lib/lua/5.2 is that you need to know the lua version.

lornat75, 2015-05-22:
I confirm that it works if yarp.so is installed in lib/lua/<version> 

The problem is how to tell cmake about the lua vesion...

traversaro, 2015-05-22:
Both `FindLua.cmake` and `FindLua51.cmake` (we are using the latter) define `LUA_VERSION_STRING`.

lornat75, 2015-05-22:
Works on Ubuntu 14.04

Tobias-Fischer, 2015-12-30:
Closing, thanks @lornat75 and @traversaro!

drdanz, 2016-01-04:
Reopening this until all the bindings are installed in the proper place.

PR #1372 (2017-11-13): Change bindings structure and JAVA bindings
This PR mainly focuses on refactoring the JAVA bindings.
While I am doing this I am also refactoring the file system organisation of the bindings folder.
Regarding the latter, the idea is to have all the language related stuff in a single folder.
Each folder than has an `examples` subfolder, a `tests` subfolders and a main `CMakeLists.txt` responsible of generating the bindings.

Regarding the JAVA, the idea is to generate (and install) two single files:

- a JAR file containing all the .class files 
- library file, containing the JNI library.

Users [... cut]