robotology/yarp #487: Namespace in thrift: generate header files in namespace-like location

francesco-romano, 2015-05-05:
When we generate code from a thrift file which defines a namespace, the resulting class is correctly enclosed in the namespace definition.
On the file system part, instead, the headers hierarchy is "flat": headers are added in the `include` folders. 
It would be nice if a proper hierarchy is added  in the file system. Unfortunately I know that there is no a standard way to do this. This example should explain better the issue.
(files taken from https://github.com/robotology/gazebo-yarp-plugins/tree/master/thrift/clock)
We have a thrift file like the following:
```thrift
namespace yarp GazeboYarpPlugins

service ClockServer {
    /** Pause the simulation if it was running
     */
    oneway void pauseSimulation();
...
```

The class is correctly generated as the following:
```cpp
namespace GazeboYarpPlugins {
  class ClockServer;
}

class GazeboYarpPlugins::ClockServer : public yarp::os::Wire {
...
```

But the filesystem hierarchy is: `include/ClockServer.h` while I had expected `include/GazeboYarpPlugins/ClockServer.h`.
With the latter it is possible to include the header file in the following way in an external library:
```
#include <GazeboYarpPlugins/ClockServer.h>
```
thus rendering explicit the library.

francesco-romano, 2015-05-07:
Related in some way to https://github.com/robotology/yarp/issues/445

elen4, 2015-05-07:
@francesco-romano in your example, can you try putting the `clock_rpc.thrift` file inside a `GazeboYarpPlugins` subdir? The `yarp_idl_to_dir` macro should then be called like this:

```cmake
yarp_idl_to_dir(GazeboYarpPlugins/clock_rpc.thrift ${CMAKE_CURRENT_SOURCE_DIR}/autogenerated CLOCKRPC_SRC CLOCKRPC_HEADERS CLOCKRPC_INCLUDES)
```

Does it work, or is there any other downside to this?

francesco-romano, 2015-05-08:
Hi @elen4.

If I try in this way, I get everything generated in the `GazeboYarpPlugins` directories: meaning both `include/GazeboYarpPlugins/*.h` and `src/GazeboYarpPlugins/*.cpp`. 
Furthermore the `Config.cmake` file has as a name the whole filesystem structure (separated with `_`)

traversaro, 2016-08-09:
The main problem (mostly aesthetic) of the solution suggested by @elen4 is that if you have [several headers files](https://github.com/robotology/gazebo-yarp-plugins/blob/master/thrift/worldinterface/autogenerated/include) like `Color.h`, `Pose.h`, this are included in the the files that uses them (such as `WorldInterfaceServer.h`) simply as `#include <Color.h>`, so any downstream project that includes them must include them as  `WorldInterfaceServer.h`.

traversaro, 2016-08-09:
> The main problem (mostly aesthetic) of the solution suggested by @elen4 is that if you have [several headers [...]

Actually I said something stupid, I did not check the correct files. All the files are correctly included with their new location.

drdanz, 2021-02-01:
I think this has been fixed for a while now. Please reopen if it is not.

closed 2021-02-01 by drdanz