robotology/yarp #489: Self-assignment potentially not handled throughout yarp

alecive, 2015-05-12:
While valgrinding some of my modules, I encountered this error:
```
==24390== Thread 5:
==24390== Source and destination overlap in memcpy(0xa15ed40, 0xa15ed40, 24)
==24390==    at 0x4C2F71C: memcpy@@GLIBC_2.14 (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24390==    by 0x5255BF0: yarp::sig::Vector::operator=(yarp::sig::Vector const&) (in /usr/local/lib/libYARP_sig.so.2.3.63.7)
...etc
```
Issue is that `yarp::sig::Vector` does not handle self assignment (https://isocpp.org/wiki/faq/assignment-operators#self-assignment-what - thanks @traversaro @francesco-romano ). And this, as specified in http://pubs.opengroup.org/onlinepubs/7908799/xsh/memcpy.html , might be dangerous. 

Now, I'm in the process of committing the proper changes to `yarp::sig::Vector` and `yarp::sig::Matrix`(by simply adding `if (this == &r) return *this;` to their copy operator), but the problem might be elsewhere (this is the reason for which I opened this issue).

pattacini, 2015-05-12:
A quick search over yarp on how `memcpy` is used didn't reveal to me any other similar pitfall, but I might have overlooked something...

alecive, 2015-05-12:
Good. Shall I close this?

lornat75, 2015-05-12:
Thanks @alecive fine with me.

alecive, 2015-05-12:
Ok. Closing.

closed 2015-05-12 by alecive