robotology/yarp #491: [getPid()][setPid()] not working in simulation

rlober, 2015-05-21:
@barbalberto @randaz81 

I am trying to get (and set) the PID gains for each of the joints on the icub for tuning purposes. Right now my issue is that when I try to read the PID gains for a particular joint the `getPid()` function never works. For reference, I am trying this with iCub_Fixed in Gazebo (I have also tried with iCub_SIM but it told me that PidRaw was not implemented).  My questions are:
-  Can I even call `get`/`setPid()` on the icub simulators? 
-  If the answer is yes, then what did I do wrong in the code snippets below?

In the following code I open a Poly driver/Pid interface for each of the robot parts, that is why I am using vectors.
**thread.h**
```cpp
std::vector<PolyDriver*>        robotDevice;
std::vector<IPidControl*>       iPids;

double Kp, Kd, Ki;
```
In the implementation I first open the drivers then `view()` the interfaces. Once those are ready I try to call `getPid()` for a single interface and a single joint.

**thread.cpp**
```cpp
/* initialization */
// numRobotParts = number of icub parts (i.e. head torso etc.) 6 in this case
iPids.resize(numRobotParts);

for(int rp=0; rp<numRobotParts; rp++)
{
    PolyDriver *tmp_robotDevice = new PolyDriver(options);
    if (!tmp_robotDevice->isValid()) {
        printf("Device not available.  Here are the known devices:\n");
        printf("%s", Drivers::factory().toString().c_str());
        return false;
    }
    robotDevice.push_back(tmp_robotDevice);

    if (!robotDevice[rp]->view(iPids[rp])) {
        printf("Problems acquiring interfaces\n");
        return false;
    }
}

/* get the PID gains for a particular joint */
// partIndex = the index of the icub part we are looking at e.g. head == 0
// jointIndex = the joint index of said part for which we want to get the Pid

Pid* currentPid;
while(!iPids[partIndex]->getPid(jointIndex, currentPid))
{
    Time::delay(0.01);
    std::cout << "waiting for gains..." << std::endl;
}
 Kp = currentPid->kp;
 Kd = currentPid->kd;
 Ki = currentPid->ki;
```
When I run this code the `getPid()` function never returns true so I get non-stop `waiting for gains...` in my terminal output. Similarly, for the `setPid()` function I have the same results.

Any ideas where I am going wrong?

Thanks in advance,
Ryan

traversaro, 2015-05-21:
For the `gazebo-yarp-plugins` part: the problem is that the `iPidControl`interface is not implemented in the `GazeboYarpControlBoardDriver` driver. Implementing it is a matter of implementing a few functions in the existing class. 
Relevant issues:
* https://github.com/robotology/gazebo-yarp-plugins/issues/154
* https://github.com/robotology/gazebo-yarp-plugins/issues/60
* https://github.com/robotology/gazebo-yarp-plugins/issues/119

barbalberto, 2015-05-21:
You code looks fine to me, I can confirm that neither of the 2 simulator is implementing get/set PID functions.
Maybe we shoul make gazebo also verbose on this like the iCub_SIM is, at least people does not get confused.

For a quick confirmation you can see that also yarpmotorgui does not show PID when used with gazebo simulator.

You can try it with the robot, asking PID is not a dangerous operation. It is however a blocking RPC call so the while loop is not needed in my opinion. When the getPid function returns either you have the correct values or you don't. There is no timing issues here to wait for.

rlober, 2015-05-21:
Thanks for the replies guys! I feel less insane now.

@traversaro 
If I wanted to start implementing `iPidControl` in `gazebo-yarp-plugins` , where should look first? If it is way too much work then I might avoid it but if it can be done in a couple of days to get something working then I might take a whack at it. 

@barbalberto 
You were right with the yarpmotorgui suggestion as expected it shows 0's in pretty much every gain. I will get rid of those while loops and give it a shot on the robot.

**for both of you and anyone else**
I also plan on using `IVelocityControl2` and `ITorqueControl` to tune those PID's as well. I am assuming that these too are not implemented in `gazebo-yarp-plugins`?

traversaro, 2015-05-21:
@rlober 
 * On how to implement IPidControl, see reply on the relative issue : https://github.com/robotology/gazebo-yarp-plugins/issues/60#issuecomment-104321019 
 * `IVelocityControl2` was implemented by @francesco-romano with also the PID setters/getters : you can check it at https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriverVelocityControl.cpp . It is also a good hint on how to implement the `IPidControl` interface methods.
* `ITorqueControl` interface is implemented in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriverTorqueControl.cpp , but no PID setters/getters are implemented. This is due to the fact that there is no Torque PID on Gazebo, because the torque is given as an input to multibody system. To read more about this, please check: 
  * https://github.com/robotology/gazebo-yarp-plugins/issues/192#issuecomment-101749089 (section Modify behavior of Torque control ) and https://bitbucket.org/osrf/gazebo/issue/1321/proper-joint-torque-feedback .

rlober, 2015-05-22:
@traversaro 

I will try implementing some of the position Pid code based on @francesco-romano 's work with velocity control. If I get it working I may just implement dummy functions for the torque pids so I can at least test that my code is fully functional. I will keep you posted on my progress. Thanks for the info.

BTW, your post on the gazebo bitbucket is pretty old now. You may want to email someone directly and be like, "Yo! Don Silvio of the IIT family wants this feature added! Do it or else iCub may have to pay your family a visit..." (with a heavy Italian accent of course).

rlober, 2015-05-27:
Ok some updates...

The [`pidTuner`](https://github.com/rlober/pidTuner) I have been working on is nearly finished. 
Here is a screen shot:
[image]

You can switch between individual joints, play with the excitation signal, see the response, tweak the pids etc. As you work it saves the gains you tune to an xml file. Here is an example of some output:
```xml
<?xml version="1.0" ?>
<!--
pidTuner Data Log
Last Updated: 2015-05-27.04:53:02 PM
-->
<part name="5-right_leg">
    <joint name="5-r_ankle_roll">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
    <joint name="4-r_ankle_pitch">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
    <joint name="3-r_knee">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
    <joint name="2-r_hip_yaw">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
</part>
<part name="0-head">
    <joint name="2-neck_yaw">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
    <joint name="1-neck_roll">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
    <joint name="0-neck_pitch">
        <mode name="position" Kp="0" Kd="0" Ki="0" />
    </joint>
</part>
```
I have it working in both position and torque control. 
[image]
[image]

However I am having some issues in velocity
[image]

More specifically `IVelocityControl2` `velocityMove()`. I invoke it for a single joint using:
```cpp
iCtrl[partIndex]->setControlMode(jointIndex, VOCAB_CM_VELOCITY);
// partIndex is an int indicating the part of iCub we are on (left_leg for instance)
// jointIndex is an int representing the joint on that part we are testing
// iVel[partIndex] is a IVelocityControl2 object pointer
// cmd is a double representing the reference velocity we want to send (the blue lines in the plots)
iVel[partIndex]->velocityMove(1, &jointIndex, &cmd);
```
When I run this command the joint goes into velocity control but it seems not to follow the comand `cmd` sent...
[image]

Any ideas if I am using `velocityMove` properly?

Finally I have noticed that, at least in Gazebo, there are some static position errors on some of the larger joints:
[image]

Any idea why this might be happening?

Thanks in advance to anyone who can help out!

Ryan

lornat75, 2015-05-27:
Great work! Will be very useful to all. Would you mind moving the repository to robotology so it gets better visibility? 

About the question on velocity, maybe a scale problem?

rlober, 2015-05-28:
@lornat75 

Thanks man! I am making a few changes then I will migrate it to `robotology`.  Should I move it to `robotology` or `robotology-playground`?

I tested the velocity with a step amplitude of 100 deg/sec but didn't get anything. It may have something to do with the fact that the pid interfaces have been implemented in `gazebo-yarp-plugins`...

randaz81, 2015-05-29:
> IVelocityControl2 was implemented by @francesco-romano with also the PID setters/getters : you can check it at https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriverVelocityControl.cpp . 

@francesco-romano
The goal of the simulator is to allow the user to test his application in an environment which faithfully reproduces the behavior of the real robot. The methods setVelPids() and getVelPids() are useful yarp methods and they will be certainly added *soon* to the real iCub. Unfortunately, they are currently not implemented by our hardware(*): indeed, on the real robot, position PIDs are used in both the position and the velocity control modes. The commanded velocity are integrated to generate a position reference which is tracked by the position PID. I think that in this case it's important to add a warning in the simulator to inform the user that this specific feature produces a behavior which is substantially different on the real robot (set/getVelPid() will always return zero causing a failure of rlober's application...). Additionally, I suggest (for the future) to add a test case in the icub-tests suite for the automatic verification of these methods. 

(*) only iCub with mechanical revision > 2.0 are equipped with high-resolution optical encoders needed to perform an explicit velocity closed loop control. setVelPids() and getVelPids() will be thus implemented on ETH-based robots only.

randaz81, 2015-06-03:
* added placeholder implementation of getVelPid() / setVelPid() for embObjMotioncontrol.
* added velocity pid tab in yarpmotorgui

rlober, 2015-06-03:
@randaz81 

Thank you for the help and info!

I have been struggling to get our iCubs in working order but I should be able to test the pidTuner tool tomorrow. For the time being I will test only pos and torque. Once those are working then I will look at velocity. iCubParis02 actually has fast encoders so we may be able to test velocity loops as well.

rlober, 2015-06-05:
@randaz81 

Ok! So I am finally testing on `iCubParis02` and So far most of the code works but I seem to have a problem with `IEncoders`:

More specifically any time I try to read an encoder I the read fails and I get stuck in the read loops:
```
while(!iEnc[partIndex]->getEncoders(encoders[partIndex].data()) )
        {
            std::cout << "Waiting on Encoders..." << std::endl;
            Time::delay(0.001);
        }
```
I have another loop, same code except with a more generous delay: `Time::delay(0.01);`, and it too gets stuck. Is this function implemented on the real robot? I think so because I have used similar code on `iCubParis01` and it worked... Also this has not been an issue with `icubGazeboSim`

Any ideas what I may have done wrong here? Am I missing some yarp devices on pc104 maybe? I doubt it because I am able to `view` the `iEnc` devices, but just and idea.

Thanks in advance!

randaz81, 2015-06-05:
I recently seen this issue to happen when yarp and icub-main are out-of-sync. You should update both the repos and recompile on both the local (your pc) and the remote (the pc104)

rlober, 2015-06-05:
Ok, I will make sure every build is at the same commit for the two repos, then get back to you.

rlober, 2015-06-05:
Everything is now on the same build, Unfortunately I seem to have an issue with CAN net 3 which I will have to debug with you guys next week over skype. I will send an email with regards to the CAN issue.

rlober, 2015-06-11:
While this issue is not "resolved" per se I think this issue should be closed because the implementation gaps have been identified and will be handled in the future. Additionally I think there will be some major changes to the `set`/`get` PID functions in the future: https://github.com/robotology/yarp/issues/499

closed 2015-06-11 by rlober