robotology/yarp #492: assertion failure while closing yarp module

randaz81, 2015-05-22:
I often get an error:
`[FATAL]Assertion failure (finished==true)
when closing a yarp module.`

It probably comes from line 451 of PortCore.cpp (yarp):
`// We should be finished now.`
`stateMutex.wait();`
`yAssert(finished==true);`
`stateMutex.post();`

It happens only on windows.

randaz81, 2015-05-22:
Test case:
just running a simple yarp read e.g:
yarp read ... /icubSim/head/state:o
when I press Ctrl+C I get pure vitual function call + the error mentioned above.
The first time I saw this error was about 1-2 month ago, with the introduction of shared libraries plugins in yarp (even if I can't surely tell if this is really correlated or not)

randaz81, 2015-05-26:
It was found that the environment variable YARP_FORWARD_LOG_ENABLE=1 is responsible for this issue (windows platform only)

barbalberto, 2016-11-03:
This has been addressed in [PR 942](https://github.com/robotology/yarp/pull/942)

randaz81, 2016-12-28:
Probably fixed by [PR 942](https://github.com/robotology/yarp/pull/942)

closed 2016-12-28 by randaz81