robotology/yarp #496: Bug in VectorOf<T> serialization

barbalberto, 2015-05-27:
Short description is that when vectorOf<**T**> is writing data through network in text mode, it manage all data as double, regardless of what **T** actually is.

The bug can be easily spotted here https://github.com/robotology/yarp/blob/master/src/libYARP_sig/src/Vector.cpp#L72 
where header.listTag is always set to BOTTLE_TAG_DOUBLE (and this function is called by vectorOf while writing).
```c++
 header.listTag = (BOTTLE_TAG_LIST | BOTTLE_TAG_DOUBLE);
```
I've prepared a quick test case here:
page https://gist.github.com/barbalberto/34560ad28aab03e7a7fe
git clone https://gist.github.com/34560ad28aab03e7a7fe.git

The effect of the bug is that when one or more vector of int is sent, the reader will see a BOTTLE_TAG_DOUBLE descriptor and tries to read an integer as it was a double messing things up. Different errors will show up depending if the bottle contains just a single vector or more data, however the reader get blocked.

Together with @drdanz we found a way to make things work and we implemented for integer. This can be extended for the other suppoerted types (BOTTLE_TAG_XXX), but we have to think what to do for custom types.

The proposed solution is in the branch "VectorOf_fixProposal" and it consist in extend the template to accept 2 parameters:
```c++
template<class T, int BOTTLE_TAG> class VectorOf;
```
then the write method will use the correct tag in this way
```c++
   header.listTag = (BOTTLE_TAG_LIST | getBottleTag());
```

This part is quite easy, the tricky one is to set the correct BOTTLE_TAG to be used, whitout having to rewrite the user code.

The solution for this is to take advantage of the 'specialization' of templates, therefore we defined a generic struct
```c++
template<class T>
struct DefaultBottleTag
{
  // right now the default value is double so it'll fall back on previous behaviour
  static const int bottleTag = BOTTLE_TAG_DOUBLE;   
};
```

BUT, for types handled by the bottle, we'll set the correct TAG specializing the default value like this:
```c++
template<>
struct DefaultBottleTag<int>
{
  static const int bottleTag = BOTTLE_TAG_INT;
};

template<>
struct DefaultBottleTag<double>
{
  static const int bottleTag = BOTTLE_TAG_DOUBLE;
};
```

The constructor of the VectorOff class will be the following:
```c++
template<class T, int BOTTLE_TAG = DefaultBottleTag<T>::bottleTag>
class yarp::sig::VectorOf : public VectorBase
{
...
```

In this way, for T=int, the DefaultBottleTag will use the specialized version and the bottleTag will be BOTTLE_TAG_INT;
This can be extended to string and int64 but I don't know how to do it for other types like vocab, list, dict... From Bottle.h

```c++
#define BOTTLE_TAG_INT 1
#define BOTTLE_TAG_VOCAB (1+8)
#define BOTTLE_TAG_DOUBLE (2+8)
#define BOTTLE_TAG_STRING (4)
#define BOTTLE_TAG_BLOB (4+8)
#define BOTTLE_TAG_INT64 (1+16)
#define BOTTLE_TAG_LIST 256
#define BOTTLE_TAG_DICT 512
```

Also custom types that cannot be matched with this types should fall back to double or maybe blob? How are blob printed on terminal from a yarp read ?
Maybe all stuff which is not easily printable can be handled as a blob, and a blob will just print "I'm a blob, therefore I cannot be printed?"

drdanz, 2015-05-28:
Thanks @barbalberto for the very detailed explaination.

I'm still not sure about what will happen if we create a vector of something different, for example:

 * A 16 bit integer `yarp::sig::vector<yarp::os::NetInt16>` (we might add a BOTTLE_TAG_INT16 tag, and handle it in the bottle)
 * An unsigned int `yarp::sig::vector<yarp::os::NetUInt32>` (we should probably add a BOTTLE_TAG_UNSIGNED that can be sent as BOTTLE_TAG_UNSIGNED | BOTTLE_TAG_INT16
 * A struct or a class `yarp::sig::vector<some_struct>` (probably serializing it as blob would be better)

@barbalberto It would be really useful if you could change your test to be a unit test in yarp, and if you could test at least all the "Net" types, plus some struct

lornat75, 2015-06-01:
To simplify and remove the struct DefaultBottleTag<> can't we directly specialize all the templates for the couples <type, bottle_tag_types> ?

drdanz, 2015-06-03:
@lornat75 I think we could add a pure virtual method `yarp::os::VectorBase::getBottleTag()` and overload it in `yarp::os::Vector<T>::getBottleTag()` and specialize it for known types (e.g. `yarp::os::vector<int>::getBottleTag()` . Then in `yarp::os::VectorBase::read()` and `yarp::os::VectorBase::write()` we can call getBottleTag() instead of `BOTTLE_TAG_LIST + BOTTLE_TAG_DOUBLE` (or just instead of `BOTTLE_TAG_DOUBLE`?)

barbalberto, 2015-06-26:
Any news about this?

lornat75, 2015-09-02:
I have revised this design and simplified (I think) a lot. I pushed the changes here:

https://github.com/lornat75/yarp/tree/VectorOf_fixProposal2

closed 2016-01-19 by lornat75