robotology/yarp #498: [robotInterface]

rlober, 2015-06-01:
I just pulled and built yarp and when I try to launch `robotInterface` I get the following errors and `robotInterface` quits.

**yarplogger errors:**
```
4974		6637.215609		ERROR		One or more devices failed opening... see previous log messages for more info 
4975		6637.679510		ERROR		Error in startup phase... see previous messages for more info 
4980		6637.679708		ERROR		park device do not exists 
4981		6637.679735		ERROR		Cannot run park action on device head_calibrator 
4982		6637.679761		ERROR		park device do not exists 
4983		6637.679787		ERROR		Cannot run park action on device torso_calibrator 
4984		6637.679813		ERROR		park device do not exists 
4985		6637.679840		ERROR		Cannot run park action on device right_leg_calibrator 
4986		6637.679866		ERROR		park device do not exists 
4987		6637.679891		ERROR		Cannot run park action on device left_leg_calibrator 
4988		6637.679916		ERROR		park device do not exists 
4989		6637.679940		ERROR		Cannot run park action on device left_arm_calibrator 
4990		6637.679965		ERROR		park device do not exists 
4991		6637.689568		ERROR		Cannot run park action on device right_arm_calibrator 
4992		6637.689620		ERROR		park device do not exists 
4993		6637.689650		ERROR		Cannot run park action on device left_hand_calibrator 
4994		6637.689678		ERROR		park device do not exists 
4995		6637.689705		ERROR		Cannot run park action on device right_hand_calibrator 
5000		6637.689846		ERROR		Error in interrupt1 phase... see previous messages for more info 
5003		6637.689920		ERROR		left_arm_cartesian is not a wrapper, therefore it cannot have detach actions 
5004		6637.689946		ERROR		Cannot run detach action on device left_arm_cartesian 
5005		6637.689971		ERROR		right_arm_cartesian is not a wrapper, therefore it cannot have detach actions 
5006		6637.689998		ERROR		Cannot run detach action on device right_arm_cartesian 
5010		6637.690098		ERROR		left_arm_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5011		6637.690124		ERROR		Cannot run detach action on device left_arm_mc_wrapper 
5012		6637.699626		ERROR		right_arm_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5013		6637.699679		ERROR		Cannot run detach action on device right_arm_mc_wrapper 
5014		6637.699710		ERROR		left_leg_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5015		6637.699736		ERROR		Cannot run detach action on device left_leg_mc_wrapper 
5016		6637.699762		ERROR		right_leg_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5017		6637.699787		ERROR		Cannot run detach action on device right_leg_mc_wrapper 
5018		6637.699813		ERROR		head_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5019		6637.699839		ERROR		Cannot run detach action on device head_mc_wrapper 
5020		6637.699864		ERROR		torso_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
5021		6637.699890		ERROR		Cannot run detach action on device torso_mc_wrapper 
5022		6637.699917		ERROR		left_arm_VFTserver is not a wrapper, therefore it cannot have detach actions 
5023		6637.699944		ERROR		Cannot run detach action on device left_arm_VFTserver 
5024		6637.699980		ERROR		left_leg_VFTserver is not a wrapper, therefore it cannot have detach actions 
5025		6637.700005		ERROR		Cannot run detach action on device left_leg_VFTserver 
5026		6637.700030		ERROR		right_arm_VFTserver is not a wrapper, therefore it cannot have detach actions 
5027		6637.700055		ERROR		Cannot run detach action on device right_arm_VFTserver 
5028		6637.700080		ERROR		right_leg_VFTserver is not a wrapper, therefore it cannot have detach actions 
5029		6637.700106		ERROR		Cannot run detach action on device right_leg_VFTserver 
5030		6637.700131		ERROR		torso_VFTserver is not a wrapper, therefore it cannot have detach actions 
5031		6637.700156		ERROR		Cannot run detach action on device torso_VFTserver 
5032		6637.700182		ERROR		left_arm_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5033		6637.700208		ERROR		Cannot run detach action on device left_arm_as_wrapper 
5034		6637.700232		ERROR		left_leg_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5035		6637.709683		ERROR		Cannot run detach action on device left_leg_as_wrapper 
5036		6637.709726		ERROR		right_arm_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5037		6637.709755		ERROR		Cannot run detach action on device right_arm_as_wrapper 
5038		6637.709781		ERROR		right_leg_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5039		6637.709807		ERROR		Cannot run detach action on device right_leg_as_wrapper 
5040		6637.709833		ERROR		left_hand_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5041		6637.709858		ERROR		Cannot run detach action on device left_hand_as_wrapper 
5042		6637.709885		ERROR		right_hand_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5043		6637.709912		ERROR		Cannot run detach action on device right_hand_as_wrapper 
5044		6637.709937		ERROR		left_foot_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5045		6637.709962		ERROR		Cannot run detach action on device left_foot_as_wrapper 
5046		6637.709988		ERROR		right_foot_as_wrapper is not a wrapper, therefore it cannot have detach actions 
5047		6637.710013		ERROR		Cannot run detach action on device right_foot_as_wrapper 
5048		6637.710039		ERROR		left_arm_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
5049		6637.710064		ERROR		Cannot run detach action on device left_arm_skin_wrapper 
5050		6637.710089		ERROR		right_arm_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
5051		6637.710116		ERROR		Cannot run detach action on device right_arm_skin_wrapper 
5052		6637.710142		ERROR		torso_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
5053		6637.710167		ERROR		Cannot run detach action on device torso_skin_wrapper 
5058		6637.719764		ERROR		Error in shutdown phase... see previous messages for more info 
```

I have the exact same problem on Paris01 and 02. I am trying to update the firmware but I am having an issue with that as well: https://github.com/robotology/icub-firmware-build/issues/1

pattacini, 2015-06-01:
Did you take https://github.com/robotology/QA/issues/33 into account?

rlober, 2015-06-01:
Here is the `bash_iCubrc` I believe my `YARP_DATA_DIRS` are set properly:

```
if [ "$PS1" ]; then 
        echo "Setting up yarp and iCub env vars"
fi

# Set the name of your robot here.
# Please change also the root user password
export YARP_ROBOT_NAME=iCubParis02

# DebugStream customization
export YARP_VERBOSE_OUTPUT=0
export YARP_COLORED_OUTPUT=1
export YARP_TRACE_ENABLE=0

export ROBOT_CODE=/usr/local/src/robot/
export ICUBcontrib_DIR=$ROBOT_CODE/iCubContrib

export YARP_ROOT=$ROBOT_CODE/yarp
export YARP_DIR=$YARP_ROOT/build-pc104
export ICUB_ROOT=$ROBOT_CODE/icub-main
export ICUB_DIR=$ICUB_ROOT/build-pc104
export icub_firmware_shared_ROOT=$ROBOT_CODE/icub-firmware-shared
export icub_firmware_shared_DIR=$icub_firmware_shared_ROOT/build-pc104

export YARP_DATA_DIRS=$YARP_DIR/share/yarp/:$ICUB_DIR/share/iCub/

export YARP_FORWARD_LOG_ENABLE=1  

# For compiling
DRIVER_ROOT=/lib/modules/$(uname -r)/iCubDrivers/
# to be used if the corresponding hardware is mounted on the robot 
#export ESDCANAPI_DIR=$DRIVER_ROOT/esdCanApi/
#export PLXCANAPI_DIR=$DRIVER_ROOT/plxCanApi/
export CFW2CANAPI_DIR=$DRIVER_ROOT/cfw002/

# Set-up optimizations
export CMAKE_BUILD_TYPE=Release

export PATH=$PATH:$ICUB_DIR/bin:$YARP_DIR/bin

# To enable tab completion on yarp port names
if [ -f $YARP_ROOT/scripts/yarp_completion ]; then
  source $YARP_ROOT/scripts/yarp_completion
fi
```

When I run `yarp plugin --all` I get the following:
```
icub@pc104:~$ yarp plugin --all
Runtime plugins found:

device cuda
  * ini file:       /usr/local/src/robot//yarp/build-pc104/share/yarp/plugins/cuda.ini
  * config:         cuda (type device) (name cuda) (library ycuda) (part cuda)
  Cannot find or load shared library

device dynamixelAX12Ftdi
  * ini file:       /usr/local/src/robot//yarp/build-pc104/share/yarp/plugins/dynamixelAX12Ftdi.ini
  * config:         dynamixelAX12Ftdi (type device) (name dynamixelAX12Ftdi) (library dynamixelAX12FtdiDriver) (part dynamixelAX12Ftdi)
  Cannot find or load shared library

device fakebot
  * ini file:       /usr/local/src/robot//yarp/build-pc104/share/yarp/plugins/fakebot.ini
  * config:         fakebot (type device) (name fakebot) (library yarp_fakebot) (part fakebot)
  Cannot find or load shared library

device ffmpeg_grabber
  * ini file:       /usr/local/src/robot//yarp/build-pc104/share/yarp/plugins/ffmpeg.ini
[... 196 lines cut ...]

device skinWrapper
  * ini file:       /usr/local/src/robot//icub-main/build-pc104/share/iCub/plugins/skinWrapper.ini
  * config:         skinWrapper (type device) (name skinWrapper) (library skinWrapper)
  Cannot find or load shared library

device skinprototype
  * ini file:       /usr/local/src/robot//icub-main/build-pc104/share/iCub/plugins/skinprototype.ini
  * config:         skinprototype (type device) (name skinprototype) (library skinprototype) (wrapper skinWrapper)
  Cannot find or load shared library

device xsensmtx
  * ini file:       /usr/local/src/robot//icub-main/build-pc104/share/iCub/plugins/xsensmtx.ini
  * config:         xsensmtx (type device) (name xsensmtx) (library xsensmtx)
  Cannot find or load shared library

```

It seems like quite a few modules are missing, but it does seem to find the ones that are explicitly activated in CMake. One that stands out is:
```
device cfw2can
  * ini file:       /usr/local/src/robot//icub-main/build-pc104/share/iCub/plugins/cfw2can.ini
  * config:         cfw2can (type device) (name cfw2can) (library cfw2can)
  Cannot find or load shared library
```

barbalberto, 2015-06-01:
Please, make sure you are after commit
e35b8dfe0be938f7d46b502864d8a10b375ba308 "Fix previous commit on parametric calibrator", maybe pull again, just in case...

rlober, 2015-06-02:
@pattacini @barbalberto

Update:
Yesterday evening with Julien, we fixed this issue by changing the `icub-main` Cmake configuration param `CFW2CANAPI_LIB` from `/usr/lib/lbcfw002.so` to `/lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.so`. Note that the flag: `CFW2CANAPI_DIR` is set to: `/lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002`. Also when i run `locate *libcfw002.so` I get the following:
```
icub@pc104:~$ locate *libcfw002.so
/lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.so
/usr/lib/libcfw002.so

```

After changing this parameter that seemed to fix our problem. Today I decided to completely rebuild `yarp`, `icub-firmware-shared`, and `icub-main` on both pc104 and the cluster. While everything built fine I have unfortunately recreated my original problem, only this time changing the `CFW2CANAPI_LIB` path doesn't seem to have any effect. 

Rather than copy and paste the output of `robotInterface` I have decided to dump it to a txt file which you can download here:
https://www.dropbox.com/s/24lw2ts0z8arim6/robotInterface.txt?dl=0

pc104 is operating on `Linux pc104 3.2.0-4-686-pae #1 SMP Debian 3.2.51-1 i686` 

Additionally I updated the firmware to see if that would fix the problem. It did not:
https://github.com/robotology/icub-firmware-build/issues/1

P.S. This may be more of an `icub-main` issue, but I do not know how to migrate it.

Thanks again,
Ryan

pattacini, 2015-06-02:
Did you purge the cmake cache? The `cfw2canapi` library should be automatically found by cmake during configuration.

Regarding the latest dump:
```
yarpdev: ***ERROR*** could not find device <sharedcan>
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
```

The `sharedcan` device is not even found... did you enable its compilation within cmake options?

For the `canmotioncontrol` I'll leave the others holding the bag... :smile: 

`robotInterface` will be moved in `yarp` soon (it's a service launcher pretty much iCub independent), so I think there is no need for migrating this issue.

rlober, 2015-06-02:
@pattacini 
Good, catch. I enabled it but still no luck. I dumped the output as before. Here is a link to it.
https://www.dropbox.com/s/7mbatt03kvthm9t/robInt.txt?dl=0

I know you guys are on holidays so thanks for trying to help.

drdanz, 2015-06-03:
@rlober Looks like you are still missing some devices:

```
yarpdev: ***ERROR*** could not find device <canBusVirtualAnalogSensor>
yarpdev: ***ERROR*** could not find device <skinWrapper>
yarpdev: ***ERROR*** could not find device <canBusSkin>
```

Can you check if these are enabled in icub_main?

rlober, 2015-06-03:
@pattacini @drdanz 

Ok so good news, I enabled those three devices and `robotInterface` is now running. At first it did not run because it could see some of the devices on can net 4 but I rebooted the boards and it found them. My guess is that some of the connections on that board are a little faulty so I will check them out - but it seems that my software problems were related to the following:

1. Initially I did not completely remove the contents of `build_*` before I recompiled with the [new architecture](https://github.com/robotology/QA/issues/33), this caused new components to be built along with the old ones rather than replacing them (I believe). **Solve:** Delete the `build_*` dirs, remake them and recompile.

2. My second issue was that I was not enabling all of the necessary CMake flags in `icub-main`. In my defense I followed the [wiki](http://wiki.icub.org/wiki/Compilation_on_the_pc104), albeit blindly, and it has been so long since I reconfigured Cmake on pc104 that I had forgotten what to enable. That being said I will write down all of the flags to enable from now on :smile:. **Solve:** Don't be a dummy and enable the devices that throw an error.

Thank you guys for all the help!
Ryan

closed 2015-06-03 by rlober