robotology/yarp #499: iPidControl and IPositionControl cleanup

randaz81, 2015-06-04:
Currently we have:
* ITorqueControl interface with set/getTorquePid() method
* IVelocitycontrol interface with set/getVelPid() method
* IPositionControl interface with no methods to set the position pid.
* IPidControl with generic set/getPid() methods (which are currently operating on the position pid parameters)

Since we the number of PIDs is grown over the years, I think that some cleanup is required. My suggestion is to add set/getPositionPid() to the IPositionControl and (in the long term) to deprecate the use of IPidControl to set the position pid (the interface can be kept to operate on a generic pid, but not on the position pid).

pattacini, 2015-06-04:
Discussing with @randaz81, we think that a cleanup can be done as follows:

1. Modify the `IPidControl` interface to provide a `set/getPid(const string &type, ...)`
2. Gradually declare deprecated the specific `set/getTorquePid()`, `set/getVelPid()` ...

where the string `type` can be `"position"`, `"velocity"`, `"torque"`, `"current"` ...

This way the interface's methods will no longer change upon the addition of a new control mode, because the we can deal with that just by extending the set of allowed string types.

randaz81, 2015-06-04:
Indeed an interface to set the current loop PID is currently missing, and this solution would solve the problem without adding a new interface...

drdanz, 2015-06-04:
> where the string `type` can be `"position"`, `"velocity"`, `"torque"`, `"current"` ...

@pattacini @randaz81 Is there a reason why it should be a string and cannot be an enum? I'm not sure about how much performance is important here, but a switch/case over an enum is faster than comparing strings, and you can extend it by just adding new values.

pattacini, 2015-06-04:
`enum` is fine, even though the string could be there only at the user level, while internally we send integers (i.e. enums).

pattacini, 2015-06-11:
In the end it was agreed to implement the [`yarp::os::IRemoteVariable`](http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IRemoteVariables.html) interface (see https://github.com/robotology/yarp/pull/500).

However, this interface is intended for service usage (kind of `God mode`).

randaz81, 2015-11-12:
Actually `yarp::os::IRemoteVariable` solved the issue of providing a quick access to internal (debug) variables, typically hardware specific and thus unworthy to be added in yarp.
However I still think that a cleanup of `iPidControl` is necessary...

randaz81, 2017-05-08:
As discussed with @traversaro, to be should be implemented in yarp 2.3.70 in 2.3.72

barbalberto, 2017-05-08:
@liesrock Please keep this issue in mind should you update yarp.

drdanz, 2017-05-16:
@randaz81 Can we close this now?

randaz81, 2017-05-17:
No, https://github.com/robotology/yarp/pull/1205 is just the first of a two-parts commit.

drdanz, 2017-06-06:
@randaz81 Can we close this now that #1220 was merged?

randaz81, 2017-06-06:
Yes, task complete!

closed 2017-06-06 by randaz81