robotology/yarp #501: getPid() gets hung up occasionally and seg faults the second time I call it

rlober, 2015-06-11:
More PID fun!!!

So I am having two problems with the code snippet below:

1. When I first call this function in the thread, sometimes `getPid` gets stuck and blocks execution. I am not sure why this happens sometimes and not others but I haven't noticed any pattern with this bug - so I just relaunch until everything works... Not the best solution.

2. The big issue I am having, however, is that when I call `getPid` (and it works) I get the Pid as expected and everything works fine, but when I call it a second time (rather, when I call `updatePidInformation()` again) I get a seg fault. I know it occurs at the `getPid` call because of the `cout`s around it. My guess is that the issue is with my code and not with the function, but I cannot figure out where! Is there something inherently wrong with my usage of `getPid`?

```cpp
void CtrlThread::updatePidInformation()
{
    Pid* currentPid;

    if (iPids[partIndex]==NULL) {
        std::cout << "no iPid device" << std::endl;
    }
    else{std::cout << "trying to get PID from joint." << std::endl;}

    bool res;
    if(isPositionMode){
        std::cout << "Test 1" << std::endl;
        res = iPids[partIndex]->getPid(jointIndex, currentPid);
        std::cout << "Test 2" << std::endl;
    }
    else if (//... Do stuff with the PID values...

}
```

Thanks in advance,
Ryan

traversaro, 2015-06-16:
Hi @rlober , sorry but in the past weeks we have been relatively busy!

Some comments, hope they helps:
  1. I checked the code and apparently `getPid` is implemented in `RemoteControlBoard` as an RPC call without any timeout... it is possible that something goes wrong in the communication and the port then waits forever? @barbalberto @drdanz 
  2. `getPid` is expecting a pointer to a `Pid` object to store the data, but you are passing a pointer that was never initialized (the local variable `Pid* currentPid`). I guess a better solution is to declare a local variable `Pid myCurrentPid` and then pass its pointer to the `getPid` function:
~~~
res = iPids[partIndex]->getPid(jointIndex, &myCurrentPid);
~~~
Fore some reason during the first call to the function the `Pid* currentPid` variable is containing a valid pointer, but perhaps you are corrupting some other variable of the program.

rlober, 2015-06-16:
Hey @traversaro,

No worries I know you are still catching up after DRC. I will try these changes out tomorrow and let you know if anything improves. Thank you for the suggestions.

rlober, 2015-06-18:
@traversaro 

Your suggestion of declaring a local pid object then passing its address worked. Here is the working code:

```cpp
void CtrlThread::updatePidInformation()
{
    Pid currentPid;
    bool res;
    res = iPids[partIndex]->getPid(jointIndex, &currentPid);

    //... Do stuff with the PID values...
}
```

Thanks a bunch!

closed 2015-06-18 by rlober