robotology/yarp #503: GyarpManager - duplicated applications into the application list

alecive, 2015-06-12:
[image]

As you can see from the image above, I have currently two `PPS_with_Optical_Flow` applications into the applications list. One is placed into the `ICUBContrib` install path, whereas the other one is in `.local`. As you can see from this diff:
```
icub@icub16:~$ diff /home/icub/.local/share/yarp/applications/PPS_OptFlow.xml /usr/local/src/robot/iCubContrib/share/ICUBcontrib/applications/PPS_OptFlow.xml 
28c28
<         <node>icub12</node>
---
>         <node>icub13</node>
34c34
<         <node>icub16</node>
---
>         <node>icub15</node>
40c40
<         <node>icub16</node>
---
>         <node>icub15</node>
46c46
<         <node>icub16</node>
---
>         <node>icub15</node>
52c52
<         <node>icub16</node>
---
>         <node>icub15</node>
58c58
<         <node>icub16</node>
---
>         <node>icub15</node>
64c64
<         <node>icub16</node>
---
>         <node>icub15</node>
70c70
<         <node>icub12</node>
---
>         <node>icub13</node>
76c76
<         <node>icub12</node>
---
>         <node>icub13</node>
81c81
<         <node>icub12</node>
---
>         <node>icub13</node>
86c86
<         <node>icub12</node>
---
>         <node>icub13</node>
```
Their only difference is that the latter is customized in order to use the nodes of the bluey instead of the blackie. So why I have this duplication? Shouldn't only the `.local` application be listed into the application list?

apaikan, 2015-06-14:
@alecive  For the yarpmanager there are completely two different applications. Applications are listed by their name but are found by their xml file. We do not have any policy yet to not list duplicated files. 

This is more the system maintanance issue. A simple (and so far the correct) way is to choose the proper name for any installed specifc application: e.g., `PPS_With_optical_Flow_Bluey` if the application is the robot/cluster dependent.

alecive, 2015-06-15:
Ok, thanks @apaikan for the clarification. Closing.

closed 2015-06-15 by alecive