robotology/yarp #504: Segmentationfault within Yarp

Arikae, 2015-06-12:
Hi,

we ran into a problem within yarp when creating our own SerialPort for reading from an external device.

This is the stacktrace:

--------------------------------------------

Thread #3 [HapticGloveTele] 15858 [core: 0] (Suspended : Signal : SIGSEGV:Segmentation fault)	
	SerialPort::SerialPortImpl::HandlePosixSignal() at 0x7ffff79d182a	
	0x7ffff79d1bfd	
	<signal handler called>() at 0x7ffff6a21eb0	
	accept() at syscall-template.S:81 0x7ffff6ae59ad	
	ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const at 0x7ffff6576429	
	yarp::os::impl::SocketTwoWayStream::open() at 0x7ffff774ed22	
	yarp::os::impl::TcpFace::read() at 0x7ffff7753833	
	yarp::os::impl::PortCore::run() at 0x7ffff76dd915	
	theExecutiveBranch() at 0x7ffff775677b	
	ACE_OS_Thread_Adapter::invoke() at 0x7ffff655bb16	
	start_thread() at pthread_create.c:309 0x7ffff60580a5	
	clone() at clone.S:111 0x7ffff6ae4cfd	

--------------------------------------------

The simpllified code that leads to the failure, when executed is the following:

std::string DEVICE_NAME = "/dev/ttyACM1";
SerialPort serial(DEVICE_NAME);
serial.Open(SerialPort::BAUD_115200, SerialPort::CHAR_SIZE_8,
		SerialPort::PARITY_NONE, SerialPort::STOP_BITS_1,
		SerialPort::FLOW_CONTROL_NONE);
if (serial.IsOpen() == true) {
	printf("Serial Port is Open!\n");
} else {
	printf("Error: Could not open serial port");
}

while (true) {
	if (serial.IsDataAvailable()) {
		next_byte = serial.ReadByte();

}

Here we open a Serialport to an external device that is connected via USB but after some readings, we get the segmentation fault. 
It tells us that yarp is doing something internally with SerialPort that we might have crashed. 
Does anyone has a suggestion what the problem might be?

Regards
Lars

apaikan, 2015-06-14:
Hi Lars,
 
Are you using yarp serial device somewhere else? YARP cannot interfere with any devices or system calls of the system if they are not explicitly called from YARP. My guess is that the problem is coming from somewhere else in your code which causes memory fault.
Anyway, have you tried to use the Yarp serial device interface instead of yours? I have been using it for different small projects and it seems it is working fine!

```C++
       #include <yarp/dev/SerialInterfaces.h>
       #include <yarp/dev/PolyDriver.h>

        PolyDriver driver;
        ISerialDevice *iSerial;
        Property prop;        
        prop.put("device", "serialport");
        prop.put("comport", "/dev/ttyACM1");
        prop.put("baudrate", 115200);
        prop.put("paritymode", "NONE");
        prop.put("databits", 8);
        prop.put("readtimeoutmsec", 1000);
        prop.put("verbose", 0);

        driver.open(prop);
        if(!driver.isValid()){
            fprintf(stderr, "Error opening PolyDriver check parameters\n");
            return false;
        }

        driver.view(iSerial);  
        if(!iSerial) {
            fprintf(stderr, "Error opening serial driver. Device not available\n");
            return false;
        }

...

        char data;        
        iSerial->receiveChar(data);
...
```

Arikae, 2015-06-15:
Thanks! 
Problem is gone and works like a charm now :)

closed 2015-06-15 by Arikae