robotology/yarp #506: Matlab and Yarp: confusing installation instructions

martz21, 2015-06-12:
Hi!

Setting up Matlab to work with Yarp is still a bit complicated because the documentation is sometimes misleading or is missing some parts and it is spread all around on multiple websites (e.g. http://wiki.icub.org/wiki/Calling_yarp_from_Matlab, http://wiki.icub.org/yarpdoc/yarp_swig.html, http://wiki.icub.org/images/6/63/Yarp4Matlab.pdf).

This is what worked for me.

1) Get swig and java-jdk:
 $ apt-get install swig
 $ apt-get install openjdk-6-jdk

2) Recompile Yarp 
with YARP_COMPILE_BINDINGS and CREATE_JAVA set to ON

3) Compile bindings:
 $ cd $YARP_ROOT/bindings
 $ mkdir build && cd build
 $ cmake .
 $ make
so now there is a folder "generated_src" in $YARP_ROOT/bindings/build.

4) Matlab side:
Find the files "classpath.txt" and "librarypath.txt" by typing in a matlab command window "which classpath" (they usually are in the Matlab installation folder in toolbox/local).
Add the following lines at the end of both files
   # YARP binding
   $YARP_ROOT/bindings/build/generated_src

Now typing 'LoadYarp' in the Matlab command window gives a reply message like "Yarp library loaded and initialized"!

I hope this might help!
M.

lornat75, 2015-06-15:
Official installation instructions are:

http://wiki.icub.org/yarpdoc/yarp_swig.html

I would like to update those with your feedback and delete the other ones you point, unless somebody objects.

traversaro, 2015-09-02:
The installation instruction should have been fixed by https://github.com/robotology/yarp/pull/572 , @martz21 feel free to reopen if the instructions are still not clear.

closed 2015-09-02 by traversaro