robotology/yarp #508: Python bindings broken in python v2.7.10 (OS X)

kt10aan, 2015-06-14:
In OS X, the default Homebrew python distribution is now at 2.7.10. (I assume that the next Ubuntu release in November will have 2.7.10 also...)

The swig version is 3.0.5.

The yarp bindings compilation fails with:
```
Linking CXX shared module ../lib/python/_yarp.so
Undefined symbols for architecture x86_64:
  "_PyArg_ParseTuple", referenced from:
      _wrap_delete_SwigPyIterator(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_value(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_incr(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_decr(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_distance(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_equal(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_copy(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      ...
  "_PyArg_UnpackTuple", referenced from:
      SwigPyObject_own(_object*, _object*) in yarpPYTHON_wrap.cxx.o
  "_PyBool_FromLong", referenced from:
      SwigDirector_Thread::afterStart(bool) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_equal(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator___eq__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator___ne__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_isEqual(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_notEqual(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap___lt__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      ...
  "_PyBool_Type", referenced from:
      SwigDirector_PortReader::read(yarp::os::ConnectionReader&) in yarpPYTHON_wrap.cxx.o
      SwigDirector_Thread::threadInit() in yarpPYTHON_wrap.cxx.o
      SwigDirector_RFModule::updateModule() in yarpPYTHON_wrap.cxx.o
      SwigDirector_RFModule::configure(yarp::os::ResourceFinder&) in yarpPYTHON_wrap.cxx.o
      SwigDirector_RFModule::respond(yarp::os::Bottle const&, yarp::os::Bottle&) in yarpPYTHON_wrap.cxx.o
      SwigDirector_RFModule::attach(yarp::os::Port&) in yarpPYTHON_wrap.cxx.o
      SwigDirector_RFModule::attach(yarp::os::RpcServer&) in yarpPYTHON_wrap.cxx.o
      ...
  "_PyBuffer_Release", referenced from:
      setExternal(yarp::sig::Image*, _object*, int, int) in yarpPYTHON_wrap.cxx.o
      setExternal2(yarp::sig::Image*, _object*, int, int) in yarpPYTHON_wrap.cxx.o
      _wrap_Image_setExternal(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_Image_setExternal2(_object*, _object*) in yarpPYTHON_wrap.cxx.o
  "_PyCFunction_Type", referenced from:
      _wrap_IControlDebug_setPrintFunction(_object*, _object*) in yarpPYTHON_wrap.cxx.o
  "_PyCapsule_GetPointer", referenced from:
      SWIG_Python_DestroyModule(_object*) in yarpPYTHON_wrap.cxx.o
  "_PyCapsule_Import", referenced from:
      _init_yarp in yarpPYTHON_wrap.cxx.o
  "_PyCapsule_New", referenced from:
      _init_yarp in yarpPYTHON_wrap.cxx.o
  "_PyClass_Type", referenced from:
      SwigPyClientData_New(_object*) in yarpPYTHON_wrap.cxx.o
  "_PyDict_GetItem", referenced from:
      SWIG_Python_GetSwigThis(_object*) in yarpPYTHON_wrap.cxx.o
  "_PyDict_New", referenced from:
      SWIG_Python_NewPointerObj(_object*, void*, swig_type_info*, int) in yarpPYTHON_wrap.cxx.o
  "_PyDict_SetItem", referenced from:
      SWIG_Python_NewPointerObj(_object*, void*, swig_type_info*, int) in yarpPYTHON_wrap.cxx.o
  "_PyDict_SetItemString", referenced from:
      _init_yarp in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_INT_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_VOCAB_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_DOUBLE_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_STRING_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_BLOB_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      BOTTLE_TAG_INT64_swigconstant(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      ...
  "_PyErr_Clear", referenced from:
      SWIG_Python_ConvertPtrAndOwn(_object*, void**, swig_type_info*, int, int*) in yarpPYTHON_wrap.cxx.o
      SWIG_AsVal_double(_object*, double*) in yarpPYTHON_wrap.cxx.o
[... 425 lines cut ...]
      SWIG_Python_NewPointerObj(_object*, void*, swig_type_info*, int) in yarpPYTHON_wrap.cxx.o
      SWIG_Python_ConvertPtrAndOwn(_object*, void**, swig_type_info*, int, int*) in yarpPYTHON_wrap.cxx.o
      _init_yarp in yarpPYTHON_wrap.cxx.o
      _wrap_delete_SwigPyIterator(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_value(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_incr(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_SwigPyIterator_decr(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      ...
  "__Py_NotImplementedStruct", referenced from:
      _wrap_SwigPyIterator___sub__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap___add__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_isEqual(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap_notEqual(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap___lt__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap___le__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      _wrap___gt__(_object*, _object*) in yarpPYTHON_wrap.cxx.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/python/_yarp.so] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make: *** [all] Error 2
```

Everything compiles fine on machines (OS X and Linux) with python v2.7.9

BrutusTT, 2015-07-08:
It seems that the Python bindings are broken independent of the Python version. I just did a quick test with the native OSX (10.10) Python (2.7) and it shows the same results.
I used a fresh checkout from the master branch with an unchanged jenkins setup that worked two month ago (last successfull run was on https://github.com/robotology/yarp/commit/becdbbee597281cd459fd9c2a9e804b7f2c77e62).

BrutusTT, 2015-07-17:
The problem is the missing python library for the linker in the ../yarp/build/bindings/CMakeFiles/_yarp_python.dir/link.txt while building. 

It seems that the CMakeLists.txt is missing the line 172 link_libraries(${PYTHON_LIBRARY}) which got removed in commit https://github.com/robotology/yarp/commit/5e54c80e2e685f5ec6515b16f0be2c2dabbd7583

drdanz, 2015-07-17:
@kt10aan, @BrutusTT thanks for reporting and investigating this.
I don't have an OSX machine, can you try if it works after the latest commit (dc23d74)?

drdanz, 2015-07-17:
Oops, dc23d74 does not build try 0fb93d3 :sweat_smile:

traversaro, 2015-07-17:
cc @nunoguedelha that has a similar issue on iDynTree .

kt10aan, 2015-07-17:
@drdanz, yep, it compiles fine here :) (OS X 10.10.3 and Homebrew python 2.7.10)

kt10aan, 2015-07-17:
I will also update the YARP homebrew formula to add a ``--with-python`` option :)

drdanz, 2015-07-17:
Thanks! Closing this.

closed 2015-07-17 by drdanz