robotology/yarp #510: Homebrew YARP formula

kt10aan, 2015-06-16:
I was very glad to see the updated OS X documentation page at http://wiki.icub.org/yarpdoc/install_yarp_mac.html

@vtikha, the instructions were very neat and tidy :)

I would like to ask about the fate of the YARP Homebrew formula. Since it is not mentioned in the updated docs, is it abandoned? Any plans of updating it? (some extensive updating of the formula is needed as dependencies have changed a lot since the last version). Also, a reminder, that the YARP formula was moved by Homebrew maintainers to the homebrew/x11 tap (https://github.com/Homebrew/homebrew-x11) as it was asking for a hard X11 dependency, something that is not the case now with Qt.

lornat75, 2015-06-16:
I would like to keep the brew formula; I think @vtikha copied instructions for compiling yarp from sources, while yarp's brew formula installs from a release/snapshot (which is the equivalent of the "install from binaries" sections in the Linux and Windows instructions). @kt10aan: can you add this section yourself?

kt10aan, 2015-06-17:
I will add the section, but first a few issues to be decided:

I updated the yarp formula in my forked homebrew-x11 repository (as it was severely outdated). You can find it here: https://github.com/kt10aan/homebrew-x11/blob/master/yarp.rb

Changes include:
- Updated release version to 2.3.64 and corresponding sha1
- Removed the obsolete gtk dependencies and, as a result, the X11 dependency
- Changed some CMake flags in the old formula to reflect better the current YARP default CMake flags (e.g., no GTK, no GYarpmanager etc)
- Enabled by default the serial and serialport devices
- The formula now has three options: 
  - --with-qt5, that builds the Qt GUIS and installs qt5 as a dependency (no Qt5_DIR variable need to be set that way)
  - --with-yarprun-log, that enables yarprun_log
  - --with-opencv, that enables the opencv_grabber device and installs OpenCV as a dependency

I could add more options, like bindings for python and lua (and adding swig as optional dependency) but for now python bindings are broken in OS X (see #508).

Two questions:
- Do you think these are sane defaults and options or do you want something different (e.g., having the Qt5 guis enabled by default)?

- Since X11 is not a hard dependency anymore, the formula does not have to be in homebrew-x11. It can moved either to main homebrew or in homebrew-science. Do you agree/want that?

You can find the full diff of the changes here: https://gist.github.com/6309a80872ba33ee3a23

kt10aan, 2015-06-17:
@lornat75 , by the way, yarp can be installed with Homebrew using the repo's HEAD:
```
brew install yarp --HEAD --option1 --option2 --option3
```
if the user does not have a specific need that the available options do not provide.

Of course, the problem is that between YARP releases, dependencies, CMake flags and defaults, can change in YARP, resulting sometimes in HEAD not compiling.

vtikha, 2015-06-17:
@kt10aan ok great, thank you. To answer your questions:
* I would leave like you did, the general default values which also set Qt5 guis to off. Then the user could simply add --with-qt5. As long as this is well documented and clear it should be ok. 
* You are right, it would be clearer to move it to either homebrew, or homebrew-science. I personally would go for homebrew-science. 
* The options for the bindings would be a good add on. I know the python ones are broken for the time being but we could start off with the lua ones then update it as things gets fixed.

kt10aan, 2015-06-17:
@vtikha, thanks.
I added the ``--with-lua`` option. The nice things is that the produced ``yarp.so`` library is automatically installed in ``/usr/local/lib/lua`` with no need for settings extra ENV variables. 
I also moved the serial/serialport to an option, and not the default.
Finally, added a "Caveat" at the end of the formula, to remind the user to add ``YARP_DATA_DIRS`` to ``.bash_profile``.

This is what ``brew info yarp``` outputs now:

```
yarp: stable 2.3.64, HEAD
Yet Another Robot Platform
http://yarp.it
Not installed
From: /Users/kostas/homebrew-x11/yarp.rb
==> Dependencies
Build: pkg-config ✔, cmake ✔
Required: ace ✔, gsl ✔, sqlite ✔, readline ✔, jpeg ✔
Optional: opencv ✔, qt5 ✔, lua ✘
==> Options
--with-lua
	Build with Lua bindings
--with-opencv
	Build the opencv_grabber device
--with-qt5
	Build the Qt5 GUI applications
--with-serial
	Build the serial/serialport devices
--with-yarprun-log
	Build with yarprun_log support
--HEAD
	Install HEAD version
==> Caveats
You need to add in your ~/.bash_profile:

    export $YARP_DATA_DIRS=/usr/local/share/yarp
```

If it's ok, I will create a PR to homebrew-x11 and in the PR I will also ask them to move the formula to homebrew-science.

@vtikha, I will also create a PR for a section for the documentation page, but pleasehave a look before it gets merged, to make sure that the overall structure and logic is maintained :)

kt10aan, 2015-06-22:
The yarp formula was update in homebrew-x11 and the request for moving it to homebrew/science was made: https://github.com/Homebrew/homebrew-science/pull/2427

kt10aan, 2015-07-01:
Added support installation of yarp bash completion in homebrew-x11 and homebrew-science (https://github.com/Homebrew/homebrew-x11/pull/100).
The migration from homebrew-x11 to homebrew-science is taking sometime (they seem to have a bigger workload than homebrew-x11), so I will go ahead and create the PR here for the updated docs.

kt10aan, 2015-07-15:
The issue is resolved I think :)

closed 2015-07-15 by vtikha