robotology/yarp #511: Error compiling YARP

Ommac, 2015-06-16:
Dear all,

I have encountered an error (see at the bottom) while compiling the latest YARP version on Ubuntu 14.04 LTS with the following CMake options:

```
ALLOW_IDL_GENERATION             OFF                                                                                                                
 CBW2_USE_ROS_MSG                 OFF                                                                                                                
 CMAKE_BUILD_TYPE                 Release                                                                                                            
 CMAKE_INSTALL_PREFIX             /usr/local                                                                                                         
 CREATE_ALLEGRO                   OFF                                                                                                                
 CREATE_CHICKEN                   OFF                                                                                                                
 CREATE_CSHARP                    OFF                                                                                                                
 CREATE_DEVICE_LIBRARY_MODULES    OFF                                                                                                                
 CREATE_GUIS                      ON                                                                                                                 
 CREATE_JAVA                      OFF                                                                                                                
 CREATE_LIB_MATH                  ON                                                                                                                 
 CREATE_LUA                       ON                                                                                                                 
 CREATE_OCTAVE                    OFF                                                                                                                
 CREATE_OPTIONAL_CARRIERS         ON                                                                                                                 
 CREATE_PERL                      OFF                                                                                                                
 CREATE_PYTHON                    OFF                                                                                                                
 CREATE_RUBY                      OFF                                                                                                                
 CREATE_SHARED_LIBRARY            ON                                                                                                                 
 CREATE_TCL                       OFF                                                                                                                
 CREATE_YARPBUILDER               ON                                                                                                                 
 CREATE_YARPDATADUMPER            ON                                                                                                                 
 CREATE_YARPDATAPLAYER            ON                                                                                                                 
 CREATE_YARPLOGGER                ON                                                                                                                 
 CREATE_YARPMANAGER               ON                                                                                                                 
 CREATE_YARPMANAGER_CONSOLE       ON                                                                                                                 
 CREATE_YARPMOTORGUI              ON                                                                                                                 
 CREATE_YARPSCOPE                 ON                                                                                                                 
 CREATE_YARPVIEW                  ON                                                                                                                 
 ENABLE_DASHBOARD_SUBMIT          OFF                                                                                                                
 ENABLE_YARPRUN_LOG               OFF                                                                                                                
 ENABLE_yarpcar_bayer_carrier     ON                                                                                                                 
 ENABLE_yarpcar_human_carrier     OFF                                              
 ENABLE_yarpcar_mjpeg_carrier     OFF                                                                                                                
 ENABLE_yarpcar_mpi_carrier       OFF                                                                                                                
 ENABLE_yarpcar_mpibcast_carrie   OFF                                                                                                                
 ENABLE_yarpcar_portmonitor_car   ON                                                                                                                 
 ENABLE_yarpcar_priority_carrie   OFF                                                                                                                
 ENABLE_yarpcar_rossrv_carrier    OFF                                                                                                                
 ENABLE_yarpcar_tcpros_carrier    OFF                                                                                                                
 ENABLE_yarpcar_xmlrpc_carrier    OFF                                                                                                                
 INSTALL_WITH_RPATH               OFF                                                                                                                
 OpenCV_DIR                       /usr/share/OpenCV                                                                                                  
 Qt5_DIR                          Qt5_DIR-NOTFOUND                                                                                                   
 SWIG_DIR                         /usr/share/swig2.0                                                                                                 
 SWIG_EXECUTABLE                  /usr/bin/swig2.0                                                                                                   
 SWIG_VERSION                     2.0.11                                                                                                             
 TEST_yarpidl_rosmsg              OFF                                                                                                                
 TEST_yarpidl_thrift              OFF                                                                                                                
 USE_LIBDC1394                    ON                                                                                                                 
 YARP_COMPILE_BINDINGS            ON                                                                                                                 
 YARP_COMPILE_TESTS               ON                                                                                                                 
 YARP_USE_ATLAS                   OFF                                                                                                                
 YARP_USE_PERSISTENT_NAMESERVER   ON                                                                                                                 
 lib_dc1394                       /usr/lib/x86_64-linux-gnu/libdc1394.so                                                                             
 lib_raw1394                      /usr/lib/x86_64-linux-gnu/libraw1394.so                                                                            

```

Error:
```
kammo@kammo-Latitude-E5540:~/Repos/yarp/build$ make
[  1%] Built target YARP_priv_sqlite
[  3%] Built target YARP_priv_gtkdatabox
[... cut ...]
/home/kammo/Repos/yarp/src/libYARP_OS/include/yarp/os/PortablePair.h:23: Warning 403: Class 'yarp::os::PortablePairBase' might be abstract, no constructors generated,
/home/kammo/Repos/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34: Warning 403: Method yarp::os::Portable::read(yarp::os::ConnectionReader &) might not be implemented.
Unable to open file /home/kammo/Repos/yarp/build/bindings/yarpLUA_wrap.cxx: Permission denied
make[2]: *** [bindings/yarpLUA_wrap.cxx] Error 1
make[1]: *** [bindings/CMakeFiles/yarp_lua.dir/all] Error 2
make: *** [all] Error 2
```

traversaro, 2015-06-16:
Can you paste the output of the following command:
~~~
ls -l /home/kammo/Repos/yarp/build/bindings/yarpLUA_wrap.cxx
~~~
?

Ommac, 2015-06-16:
It has root privileges:
```
kammo@kammo-Latitude-E5540:~/Repos/icub-main/build$ ls -l /home/kammo/Repos/yarp/build/bindings/yarpLUA_wrap.cxx
-rw-r--r-- 1 root root 3365837 feb  7 14:59 /home/kammo/Repos/yarp/build/bindings/yarpLUA_wrap.cxx
```

traversaro, 2015-06-16:
I have the impression that you generated the 'yarpLUA_wrap.cxx'  as root and now you can't modify it... Did you tried in a brand new build directory?

Ommac, 2015-06-18:
Yes, I tried and it compiles now. Thanks.

closed 2015-06-18 by Ommac