robotology/yarp #512: Yarp port polling in Qt

rlober, 2015-06-16:
Hey guys,

So this is more of an open question than an issue but I figured it was specific to yarp so it would be best to pose it here.

Basically my question is: What is the best way of polling yarp ports in a Qt application? Much like in `yarpmotorgui` I would like to read some yarp ports at the same rate as a typical `RateThread` (every ~10ms). If I was working with modules and threads I would just do something like:
```
void thread::run(){
    BufferedPort<Bottle> p;
    ...
    Bottle *b = p.read(false);
    if (b!=NULL) {
      // data received in *b
    }
}
```
I am relatively new to Qt but ideally I would like to have a signal that is emitted or an event which is triggered when a `BufferedPort` reads a non-NULL bottle. 

I have looked through the various Qt based applications but I can't seem to locate chunk of code to point me in the right direction.

Any suggestions or help would be more than welcome! Thanks in advance,
Ryan

alecive, 2015-06-17:
@rlober I'm not so much into `Qt` programming, but if you want to have a gui that does something as soon as there is an event on a port, you can create your standard yarp-based `C++` code (with maybe a callback on that port) and then "link" it to the `Qt` part. I suggest you to keep tinkering with the Qt-GUIs available in the repositories in order to understand how they do that. But consider that:
 * which GUI to play with heavily depends on the application you're thinking at. I'd suggest the `yarpscope-qt` -> The plotter class seems to do the job there https://github.com/robotology/yarp/blob/master/src/yarpscope-qt/plugin/plotter.h
 * I cannot provide you more than this feedback :)

francesco-romano, 2015-06-17:
Some notes on general GUI programming:

- you want your GUI to remain responsive: usually GUI code runs on the main thread, so your aim is to keep the main thread as free as possible => do processing on a background thread
- For the above reason, you have to be careful that GUI methods are called in the main thread

rlober, 2015-06-17:
@alecive @francesco-romano 

Thanks for your help guys. @alecive, the `plotter.h` code seems to be a good place to start, thank you. It looks like I need to create special classes which inherit from `QObject` and add the yarp port functionality there. I will leave the issue open for the day to see if anyone else wants to chime in  (@lornat75 @randaz81) , but I will close it at the latest tomorrow.

alecive, 2015-06-17:
Great! Later on, tell us what did you come up with, it might be of interest for other users :+1:

rlober, 2015-06-18:
Will do!

closed 2015-06-18 by rlober