robotology/yarp #514: [yarp::os::Type::~Type()] cause a segfault

EnricoMingo, 2015-06-17:
Hi, as in the title, I have recently passed to the last ```master``` version of YARP from the ```3ef253c``` version and running my tests on the interfaces for OpenSoT (where I am using the BufferedPorts) I have encountered this problem. I have noticed that in the ```3ef253c``` the class Type is missing.

traversaro, 2015-06-17:
Can you paste a backtrace of the segfault ?

EnricoMingo, 2015-06-17:
Sure, sorry, here it is:

```
#0  0x00007ffff7b6de77 in yarp::os::Type::~Type() () from /home/enrico/walkman/build/drc/OpenSoT/libOpenSoT.so.0.2.0
#1  0x0000000000428868 in yarp::os::BufferedPort<OpenSoT::interfaces::yarp::msgs::yarp_trj_msg_portable>::BufferedPort() ()
#2  0x00007ffff7af1cc9 in OpenSoT::interfaces::yarp::tasks::YCartesian::YCartesian(std::string const&, std::string const&, std::string, yarp::sig::Vector const&, iDynUtils&, std::string, std::string) ()
   from /home/enrico/walkman/build/drc/OpenSoT/libOpenSoT.so.0.2.0
#3  0x000000000041b60c in (anonymous namespace)::testYTask_testPoseTwistMsgs_Test::TestBody() ()
#4  0x00007ffff77dab8c in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#5  0x00007ffff77d5f7c in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#6  0x00007ffff77c2d5b in testing::Test::Run() () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#7  0x00007ffff77c347e in testing::TestInfo::Run() () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#8  0x00007ffff77c3a06 in testing::TestCase::Run() () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#9  0x00007ffff77c8700 in testing::internal::UnitTestImpl::RunAllTests() () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#10 0x00007ffff77dbad3 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) ()
   from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#11 0x00007ffff77d6eda in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) ()
   from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#12 0x00007ffff77c760c in testing::UnitTest::Run() () from /home/enrico/walkman/build/drc/OpenSoT/gtest/libgtest.so
#13 0x00000000004223ee in main ()
```

drdanz, 2015-07-15:
> I have noticed that in the 3ef253c the class Type is missing

@EnricoMingo can you explain better what you mean with this sentence?

pattacini, 2015-09-02:
ping @EnricoMingo

drdanz, 2015-09-08:
Closing. No feedback received in 2 months.
Please reopen and add some info if this is still a issue

closed 2015-09-08 by drdanz