robotology/yarp #518: yarp::math namespace - pinv function

vicentepedro, 2015-06-23:
Hi,

I was trying to use the pseudo inverse function within YARP as described here:
http://wiki.icub.org/yarpdoc/namespaceyarp_1_1math.html#a0ac2a96f0974e0e43bbfbc2230ce6c95

Although, the include file (SVD.h) needed  to use this function is not specified and not clear.

I had to figure it out.

I suggest to add it somewhere.

alecive, 2015-06-26:
@vicentepedro I don't get your answer, but the only thing I need to do to use this library is : 
```
#include <yarp/math/SVD.h>
using namespace yarp::math;
...
Matrix H_pinv = pinv(H);
```

pattacini, 2015-06-26:
@vicentepedro struggled while he was wandering through `yarp` to seek for the corresponding header file.

This issue must be solved not at the level of the single method documentation but rather by means of proper `doxygen` tags to group up routines defined in the same module.

vicentepedro, 2015-06-26:
@alecive, @pattacini is right. I posted this issue just as a suggestion for the yarp documentation. (in this case the pseudo inverse function in the yarp::math namespace documentation)

closed 2015-09-02 by lornat75