robotology/yarp #520: yarpmotorgui-qt issues

alecive, 2015-06-25:
There are still some missing issues on the yarpmotorgui that needs to be addressed in order for it to really replace its gtk counterpart.
 * [ ] it takes way longer for it to load: if the user chooses to use all the body parts, I think that it takes at least 2-3 seconds more than the gtk version. This might seem a minor detail, but it is not when you start doing live tv shows :)
 * [ ] the textbox that encloses the actual encoder value more often than not is visually cut out, and it is impossible for the user to get precise information about the joint angle
 * [ ] the slider that controls the joint angle does not round the encoder measure to its closest integer, but it does it to the _lowest_ (i.e. `4.9` becomes `5.0`)

randaz81, 2015-06-25:
I experienced some problems using the sequence tab, i.e. sometimes the commands are not sent to robot, even if the cycle animation seems to work correctly (the current executed line becomes green).

randaz81, 2015-10-22:
The above issue was a bug recently fixed by aitek

alecive, 2015-10-22:
Regarding the first item, it is a wontfix. Apparently, the `yarpmotorgui` gets a ~200ms delay when calling this `Qt` method: https://github.com/robotology/yarp/blob/83466b733d89d2f3687542b047bcd5185d3f9299/src%2Fyarpmotorgui-qt%2Fmainwindow.cpp#L356

drdanz, 2016-08-19:
@randaz81, @alecive  Is there any progress on this?