robotology/yarp #521: Qt5 guis have no menu on Ubuntu 14.04

drdanz, 2015-06-25:
On Ubuntu 14.04 all Qt5 guis do not have the menu.

Qml guis print this warning:

```
appmenu-qt: handleReparent 128 The given QWindow has no QMenuBar assigned
I would send a position, but there's no image for scaling
```

"Normal" Qt5 guis do not give any warning.

Apparently this is a bug in appmenu-qt5. A workaround to fix this issue is:

```
sudo apt-get remove --purge appmenu-qt5
```

alecive, 2015-06-25:
@drdanz and what happens when you purge `appmenu-qt5`?

vtikha, 2015-06-25:
@drdanz , quick question, does the menu reappear if you switch between apps opened and return to the qt one?

drdanz, 2015-06-25:
No, it is always missing.
When you remove the appmenu-qt5 package the menu is shown "normally" on the top of the application window and not in the top bar (Os X style- like). Anyway in my opinion, not having the menu is a much bigger issue than having the menu in the window instead of in the top bar.

drdanz, 2015-06-25:
Closing this since the documentation is now updated to explain this workarond. I will open a fake QA question.

closed 2015-06-25 by drdanz

serena-ivaldi, 2018-08-21:
Hi I am reopening this issue because the problem appeared again on our machine (Ubuntu 16). 

We recently updated yarp and now when launching yarpview we have the error:
``icub@icub01:~/software/src/yarp/build$ yarpview
appmenu-qt: handleReparent 138 The given QWindow has no QMenuBar assigned
Segmentation fault (core dumped)``

I tried to use the workaround by purging appmenu-qt and appmenu-qt5 but the only effect is the disappearing of the handleReparent error message, it still get a segmentation fault.

serena-ivaldi, 2018-08-21:
By the way, I tried the following options:
```
icub@icub01:~$ yarpview
appmenu-qt: handleReparent 138 The given QWindow has no QMenuBar assigned
Segmentation fault (core dumped)
icub@icub01:~$ yarpview --compact
Segmentation fault (core dumped)
icub@icub01:~$ yarpview --minimal
Segmentation fault (core dumped)
icub@icub01:~$ yarpview --keep-above
appmenu-qt: handleReparent 138 The given QWindow has no QMenuBar assigned
Segmentation fault (core dumped)
```
so we have segmentation fault also in the minimal /compact case. It is weird because the other graphic GUIs work (yarpmanager etc)

Nicogene, 2018-08-28:
Hi @serena-ivaldi, have you tried re-compiling yarp guis after purging `appmenu-qt`?
By the way I think that this an Unity issue, so once you migrate to 18.04 (that has GNOME as DE) you will have not anymore this problem.

ems7, 2018-10-05:
I am having the same issue, as @serena-ivaldi. I have tried recompling yarp after purging `appmenu-qt` and `appmenu-qt5`, but still get the seg fault when trying to run yarpview.