robotology/yarp #524: [yarpmanager] Add module/connection button

drdanz, 2015-07-02:
It would be nice to have a + button to add modules and connections

alecive, 2015-07-03:
:+1: but we should be more specific on this. What would you like to achieve?

vtikha, 2015-07-03:
How I see this is, that one will be able to extend the application xml by adding a module and its connections directly from the GUI and eventually save it without opening and editing the actual xml file. It is actually similar to the yarpbuilder functionality.

drdanz, 2015-07-03:
Yes that's what I want to achieve, even though for me it's not even necessary to be able to save the xml file.
Also another useflu functionality would be a checkbox for each module and connection to disable them from the "global" run/connect, but that's another feature request...

alecive, 2015-07-03:
A super cool one (we already talked about that with @apaikan ) would be also a `yarp clean` button and a `yarp name list` button :)

pattacini, 2015-07-03:
I'm in favor of check-boxes and whatsoever but not of the possibility to add modules from within the `yarpmanager`, since - as @vtikha alreday pointed out - it is a pertinence of `yarpbuilder`, thus I wouldn't mix the two of them.

Btw, I've heard that in the future we will have just the builder.

vtikha, 2015-07-03:
I also like what @drdanz suggested, checkboxes for each module and connections. It is quite similar to how the data is selected in the yarpdataplayer. `yarp clean` is ok, but i don't see the usefulness of `yarp name list` in the manager. How would this work? Where will it redirect its output etc? Maybe we should change the title of this issue to **[yarpmanager] Feature requests** :smile:

alecive, 2015-07-03:
@vtikha I was simply thinking of a new tab that would automatically open with a list of the ports. It is not that useful (at least not as useful as the `yarp clean`), but still: these two are two of the most used command on the command line (at least by me), and it would be helpful to save some clicks :D