robotology/yarp #529: Unify yarp-config documentation

gsaponaro, 2015-07-10:
Hi, I was checking some stuff about ```yarp-config``` in order to do some software upgrades on our robot (thanks @pattacini for the heads-up), and I noticed that this utility has two different documentation pages resulting from a quick Google search:

1. http://wiki.icub.org/yarpdoc/yarp_yarp-config.html
2. http://wiki.icub.org/wiki/Yarp-config

Suggestion: delete or clear the content of one of those pages, leaving only the other one. They have a lot of overlap, except for the "Robot configuration (on the PC104)" section which is only in the wiki page at the moment.

lornat75, 2015-07-10:
Removed redundant text in http://wiki.icub.org/wiki/Yarp-config and added link to http://wiki.icub.org/yarpdoc/yarp_yarp-config.html.

lornat75, 2015-07-20:
can we close this issue?

gsaponaro, 2015-07-20:
Yep, thanks! :)

By the way, I think it would be even better to completely **remove** the obsolete/older pages, so that they stop showing up on Google, but that's probably a separate issue. In this particular case, the pc104-specific instructions that still exist on the wiki page would have to be moved elsewhere, then that page removed.

closed 2015-07-20 by lornat75