robotology/yarp #532: Serializable string

drdanz, 2015-07-15:
It would be useful to have a string class that derives from `yarp::os::PortReader` and `yarp::os::PortWriter`, the easiest method that I found to do it is using `yarp::os::Value` but that looks redundant.

pattacini, 2015-09-02:
I think people can survive sending/receiving `yarp::os::Value` and/or `yarp::os::Bottle`.

I vote not to implement this.

apaikan, 2015-09-02:
I would also vote for not implement this for few reasons: 

1- The `Bottle (const ConstString &text)` or `Value (const ConstString &str)` is handy and simple enough for this purpose. Plus, they do not add too much overhead when serializing the string on the network.

2- It seems we have typdef for ConstString and std::string. Having a serializable (yarp portable) string means that we must stick to custom implementation of the string in Yarp.

closed 2015-09-02 by apaikan