robotology/yarp #536: fake_motor example does not start properly

plinioMoreno, 2015-07-18:
After compiling properly the fake_motor example, I got the following when executing it:

============================================================
check our device can be instantiated directly
[INFO]created device <motor>. See C++ class FakeMotor for documentation.
Direct instantiation worked
FakeMotor reporting 2 axes present
  number of axes is: 2

============================================================
check our device can be wrapped in controlboard
[INFO]ControlBoard subdevice is motor
[INFO]created device <motor>. See C++ class FakeMotor for documentation.
yarp: Port /controlboard/rpc:i active at tcp://10.0.3.31:10006
yarp: Port /controlboard/command:i active at tcp://10.0.3.31:10007
yarp: Port /controlboard/state:o active at tcp://10.0.3.31:10008
FakeMotor reporting 2 axes present
FakeMotor reporting 2 axes present
[INFO]Server control board starting
[INFO]created wrapper <controlboard>. See C++ class yarp::dev::ServerControlBoard for documentation.
Indirect instantiation worked
[INFO]Server control board stopping

============================================================
check our device can be accessed via remote_controlboard
[INFO]ControlBoard subdevice is motor
[INFO]created device <motor>. See C++ class FakeMotor for documentation.
yarp: Port /server/rpc:i active at tcp://10.0.3.31:10009
yarp: Port /server/command:i active at tcp://10.0.3.31:10010
yarp: Port /server/state:o active at tcp://10.0.3.31:10011
FakeMotor reporting 2 axes present
FakeMotor reporting 2 axes present
[INFO]Server control board starting
[INFO]created wrapper <controlboard>. See C++ class yarp::dev::ServerControlBoard for documentation.
Server instantiation worked
yarp: Port /client/rpc:o active at tcp://10.0.3.31:10012
yarp: Port /client/command:o active at tcp://10.0.3.31:10013
yarp: Port /client/state:i active at tcp://10.0.3.31:10014
yarp: Port /client/stateExt:i active at tcp://10.0.3.31:10015
yarp: Receiving input from /client/rpc:o to /server/rpc:i using tcp
yarp: Sending output from /client/rpc:o to /server/rpc:i using tcp
yarp: Receiving input from /client/command:o to /server/command:i using udp
yarp: Sending output from /client/command:o to /server/command:i using udp
yarp: Receiving input from /server/state:o to /client/state:i using udp
yarp: Sending output from /server/state:o to /client/state:i using udp
[ERROR]*** Extended port /server/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility behaviour
[WARNING]Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
[ERROR]received an unknown request after a VOCAB_GET: prot
[ERROR]expecting protocol 1 4 0, but remotecontrolboard returned protocol version 16886560 0 6
[ERROR] please update YARP or the remotecontrolboard implementation
[ERROR]checkProtocolVersion failed 
yarp: Removing output from /client/rpc:o to /server/rpc:i
yarp: Removing input from /client/rpc:o to /server/rpc:i
yarp: Removing input from /client/command:o to /server/command:i
yarp: Removing output from /client/command:o to /server/command:i
yarp: Removing input from /server/state:o to /client/state:i
yarp: Removing output from /server/state:o to /client/state:i
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
Client instantiation failed
[INFO]Server control board stopping

lornat75, 2015-07-21:
I think you are mixing different versions of remotecontrolboard and controlboard; @barbalberto can help...

plinioMoreno, 2015-07-21:
I am just running the `fake_motor` executable, I do not have anything else running besides the yarpserver.

Should I execute something else before?

barbalberto, 2015-07-21:
I guess the fake_motor example has to be updated with the some recent changes done in the yarp wrapper. I'll check and fix it asap. Thanks for reporting.

lornat75, 2015-09-02:
@barbalberto fixed this issue in c8889e495a7413ed182447c8d07d3bdcc37aa6af. Closing.

closed 2015-09-02 by lornat75