robotology/yarp #538: YARP-ROS - Publishing for the first time on a topic

plinioMoreno, 2015-07-18:
In this repository
https://github.com/plinioMoreno/yarp-ros-tests.git

I put an example that illustrates this, 

https://github.com/plinioMoreno/yarp-ros-tests/blob/master/currentFixationPointRead.cpp

The issue is that the command `rostopic echo /my_x_out` does not print anything, even though the port `yarp read ... /my_x_out@/ros_tests` prints properly the values.

My workaround was to create a [python script] (https://github.com/plinioMoreno/yarp-ros-tests/blob/master/x_out_port_ros_init.py) that is executed before the currentFixationPointRead. This means that the subscriber must exist before the publisher, isn't it? This is not the case for ROS.

barbalberto, 2015-07-20:
Hi Plinio,

the reason your test case does not work is that for ROS to be able to read a message, it does need to know exactly the data type it has to deal with.
Therefore a ROS compliant publisher has somehow to advertise the type of data it will publish so that other nodes on the network (rostopic echo as well) will know how to read the data. This part was missing in your case so ROS was not able to understand what you were publising.

This can be seen with the command "rostopic info  /my_x_out", the answer was:
```
rostopic info /my_x_out
Type: *

Publishers:
 * /ros_tests (http://10.255.72.134:10073)

Subscribers: None
```
As you see the "Type" field is empy.

The yarpidl_rosmsg executable does not help in this case, it makes yarp port to be able to dynamically interpret a ROS message coming from a ROS topic e translate it into a yarp compatible message. This is the reason why you are able to use a yarp port to read data from a ROS topic without having to do nothing in your code. In you example you can see the variable reading1Mux1 corretcly filled in.

To do the vice versa, i.e. writing data TO a ROS topic, you need a way to advertise the data you'll be publishing. 
This is exactly what the python script was doing, creating a node and specifying the datatype. After that, the ROS framwork will know how to handle the data.
In order to do this from YARP code, you'll need to generate an 'ad hoc' YARP portable as described here http://wiki.icub.org/yarpdoc/yarp_with_ros.html in the section "Generating YARP portables from ROS types".
In your case the command 
``` yarpidl_rosmsg  geometry_msgs/Point ```
 will generated an header file "geometry_msgs_Point.h" defining the portable for you.
Then you have to use a BufferedPort with this type of data; this will make ROS happy because a BufferedPort is a template that can never change the type of data during its lifecycle, so we can safely advertize this type of data (this cannot be done with a simple yarp port because in each write the data transmitted can be completely different). If you follow the documentation page, there is a link to the page  "Port Power, Going Further with Ports" explaining how to use buffered ports.

I cannot push on your repo, so I'll post the code fixed here:

``` c++

#include <geometry_msgs_Point.h>

int main(int argc, char *argv[])
{
        Network yarp;
        printf("Using buffered ports\n");
        BufferedPort<geometry_msgs_Point>  receiverBuff1Mux1;
        bool receiver1Mux1Ok = receiverBuff1Mux1.open("/my_other_port_in");
        BufferedPort<geometry_msgs_Point> outputPort;
        outputPort.setWriteOnly();
        bool outputOk_1 = outputPort.open("/my_x_out+@/ros_tests");

        if(!outputOk_1)
        {
            printf("outputOk_1 failed to open\n");
            return -1;
        }

        bool connectSuccess = false;
        while(!connectSuccess)
        {
            printf("\nTrying to connect to /iKinGazeCtrl/x:o ... ");
            connectSuccess = yarp.connect("/iKinGazeCtrl/x:o", receiverBuff1Mux1.getName());
            yarp::os::Time::delay(1);
        }

        printf("Connection done!\n");

        while(true)
        {
            geometry_msgs_Point *reading1Mux1;
            reading1Mux1 = receiverBuff1Mux1.read();

            geometry_msgs_Point & out = outputPort.prepare();
            out = *reading1Mux1;
            outputPort.write();
        }
}
```

***However far better then using BufferedPort*** is to use the following 2 objects dedicated to YARP-ROS intercommunication which are called:
```
        yarp::os::Node    *rosNode;          // Create a dedicated ROS node
        yarp::os::Publisher<ROS_msg>      rosPublisherPort;  // Dedicated ROS topic publisher
```

The first will create a ROS node, it is not actually used but will make ROS framework happier.
The yarp::os::Publisher behaves like a buffered port but it is dedicate to interoperate with ROS. It will help keeping compatibility working in future in case of changes in yarp or ros.

The usage in your case is the following:

``` c++
#include <yarp/os/Node.h>
#include <yarp/os/Publisher.h>
#include <yarp/sig/Image.h>
#include <geometry_msgs_Point.h>

using namespace yarp::os;

int main(int argc, char *argv[])
{
        Network yarp;
        printf("Using ROS specific objects\n");

        bool success = false;

        yarp::os::Node    *rosNode;                                                  // Create a dedicated ROS node
        yarp::os::Publisher<geometry_msgs_Point>        rosPublisherPort;           // Dedicated ROS topic publisher
        BufferedPort<geometry_msgs_Point>               receiverBuff1Mux1;
        bool receiver1Mux1Ok = receiverBuff1Mux1.open("/my_other_port_in");

        rosNode = new yarp::os::Node("/ros_tests");

        if(rosNode == NULL)
        {
            yError() << " opening ros_tests Node, check your yarp-ROS network configuration\n";
            success = false;
            return -1;
        }
        yInfo() << " opened ros_tests Node!\n";

        if (!rosPublisherPort.topic("/my_x_out") )
        {
            yError() << " opening /my_x_out Topic, check your yarp-ROS network configuration\n";
            success = false;
            return -1;
        }
        success = true;

        bool connectSuccess = false;
        while(!connectSuccess)
        {
            printf("\nTrying to connect to /iKinGazeCtrl/x:o ... ");
            connectSuccess = yarp.connect("/iKinGazeCtrl/x:o", receiverBuff1Mux1.getName());
            yarp::os::Time::delay(1);
        }

        printf(" done!\n");

        while(true)
        {
            geometry_msgs_Point *reading1Mux1;
            reading1Mux1 = receiverBuff1Mux1.read();

            geometry_msgs_Point out = *reading1Mux1;
            rosPublisherPort.write(out);
        }

        delete rosNode;
    }
    return 0;
}

Please, use this pattern also for the other examples you created and let me know if you have other issues.

plinioMoreno, 2015-07-20:
@barbalberto , Thanks for the explanation and the new information. The ROS topic is visible as you said.

closed 2015-07-21 by barbalberto