robotology/yarp #539: Reading from one ROS topic and writing to another ROS topic in the same code

plinioMoreno, 2015-07-18:
I have not been able to do this. The two files from this [repository](https://github.com/plinioMoreno/yarp-ros-tests) that illustrate the issue:

[File 1](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTest.cpp) and [File 2](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTestSwap.cpp).

There is a common behavior when running both files: I am able to publish to the ROS topic `/my_xd_in`, but I cannot read either with `yarp read ... /my_x_out@/ros_tests` nor with `rostopic echo /my_x_out`.

The other issue is the order of the connections between the ports. When is done in the following order:
`  yarp.connect("/iKinGazeCtrl/x:o", receiverBuff1Mux1.getName());
  yarp.connect(xd_outputPort.getName(),"/iKinGazeCtrl/xd:i");`

I get address conflicts:
`yarp: Port /ros_tests failed to activate at tcp://10.0.3.31:10071 (address conflict)
yarp: Port /my_x_out+@/ros_tests failed to activate at tcp://10.0.3.31:10073 (address conflict)`

In the swapped order, I do not get address conflicts, but still I am not able to read with `yarp read ... /my_x_out@/ros_tests` nor `rostopic echo /my_x_out`.

plinioMoreno, 2015-07-20:
@barbalberto ,

I changed the  [File 1](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTest.cpp)  and [File 2](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTestSwap.cpp) following your instructions from this [issue] (https://github.com/robotology/yarp/issues/538). Now all the ports are `BufferedPort<geometry_msgs_Point>`.

However, the issue still occurs. I'm able to write on the topic `/my_xd_in`, but I'm not able either to read from the port `/my_x_out@/ros_tests` nor the topic `/my_x_out`. This makes me wonder if the `yarp::os::Node` class just allows one topic, but this is just a guess from the behaviour.

The difference now is that I do not get the `(address conflict)`, but in any case I'm not able to read from the  `/my_x_out@/ros_tests` nor the topic `/my_x_out`.

plinioMoreno, 2015-07-20:
@barbalberto,
I added another example [here](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTestUsingNode.cpp) where I use both the `yarp::os::Publisher`and `yarp::os::Subscriber`.

In this case I cannot see any of the topics by issuing `rostopic list`

barbalberto, 2015-07-21:
Hi Plinio,

well there are 3 reason why the example does not work.

1) The rosNode is the 'fake' ROS executable that will publish the topics, therefore it is the first thing that needs to be created, **before** creating any topics.
This is because the node name needs to be added as a decoration to the port/topic name. If no node name is defined, the port will be opened as a standard yarp port so it will not registered as a ROS topic.

2) yarp::os::Subscriber and yarp::os::Publisher has the specific method 'topic' to open a ROS topic. 
```   bool outputOk = xd_inputPort.topic("/my_xd_in"); ```
This will add the necessary decoration to the port name, i.e. the node name and other special characters. Alternatively you can use the open method creating the full name by yourself, but using the specific method will help with future compatibility changes.

3) I run the example using start_common script and then started the ./readAndWriteTopicsTestUsingNode executable. I don't know if I'm missing some other steps.
The example program is supposed to read from a topic called /my_xd_in some data but in this scenario no one was creating this topic, so the main was stuck in the read, since no data was ever received.
Creating a topic broadcasting data with the command:
```  rostopic pub /my_xd_in geometry_msgs/Point 1 2 3 -r 1 ```
the input topic starts receiving data and the desired output is sent through the /my_x_out topic.

From the difficults you are facing, probably the documentation is not clear enough and should be revised. Maybe can we use your file as an example?

plinioMoreno, 2015-07-21:
Alberto,

I was missing something in my example ( a bit rusty on the YARP Port stuff ), but now both examples are working properly ([File that does not use yarp::os::Node](https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTest.cpp) and [File that uses yarp::os::Node] (https://github.com/plinioMoreno/yarp-ros-tests/blob/master/ReadAndWriteTopicsTestUsingNode.cpp). Now, one port is always sending the current icub gaze position and the other port is waiting for user input (the target gaze point).

From your explanation I got what I was missing in the code. I had to open one of the port using the `read(false)` to avoid the blocking when calling the part that writes onto the topic.

Thank you.

closed 2015-07-21 by plinioMoreno