robotology/yarp #548: Cannot find YARP without languages enabled

drdanz, 2015-07-22:
When CMake config files are installed in ``/usr/lib/<arch>/YARP`` (i.e. on Debian/Ubuntu when prefix is `/usr` like for system packages) projects that use ``project(foo NONE)`` cannot find YARP because the architecture is not defined.

for example, this will fail:

```cmake
cmake_minimum_required(2.8.9)
project(test NONE)
find_package(YARP REQUIRED)
```

pattacini, 2015-07-22:
@drdanz is it a real bug?

I mean, if we don't provide cmake with the info about the architecture, how can it find the correct location where to seek for the installed yarp?

Thus, why don't we restore the old way in `icub-contrib-common` removing the `none` attribute?

drdanz, 2015-07-23:
Well... the icub-contrib-common repository is already a hack, so another hack is a good fix.
Anyway I was worried about superbuilds (using YCM), that will not find YARP if they use NONE.
Perhaps just using ``enable_language(C)`` could be enough, I'll do some tests.