robotology/yarp #563: thrift_location_not_found on a freshly installed new Ubuntu 14.04

serena-ivaldi, 2015-08-14:
Hi, I just installed a new Ubuntu 14.04 machine, and I am installing everything from scratch.
During the cmake of YARP if I try to enable allow_idl_generation I currently get this error:

CMake Error at conf/YarpIDL.cmake:95 (message):
   yarpidl_thrift (YARPIDL_thrift_LOCATION-NOTFOUND) failed, aborting.
 Call Stack (most recent call first):
   src/yarpdataplayer-qt/CMakeLists.txt:57 (yarp_idl_to_dir)

any clue?

traversaro, 2015-08-14:
Can you paste the content of your `CMakeCache.txt` file? Thanks!

serena-ivaldi, 2015-08-14:
# This is the CMakeCache file.
# For build in directory: /home/icub/software/src/yarp/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a library.
ACE_ACE_LIBRARY_DEBUG:FILEPATH=ACE_ACE_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_ACE_LIBRARY_RELEASE:FILEPATH=/usr/lib/libACE.so

//Additional compiler flags for ACE
ACE_DEFINITIONS:STRING=

//Path to a library.
ACE_DL_LIBRARY_DEBUG:FILEPATH=ACE_DL_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_DL_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libdl.so

//ACE include directory
ACE_INCLUDE_DIRS:PATH=

//Path to a library.
ACE_RT_LIBRARY_DEBUG:FILEPATH=ACE_RT_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_RT_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/librt.so

//Allow YARP to (re)build IDL files as needed
ALLOW_IDL_GENERATION:BOOL=ON

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: Release MinSizeRel Debug
// RelWithDebInfo DebugFull Profile.
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler.
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during debugfull builds.
CMAKE_CXX_FLAGS_DEBUGFULL:STRING=-O0 -g3 -ggdb -fno-inline -DDEBUG

//Flags used by the compiler during release minsize builds.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during profile builds.
CMAKE_CXX_FLAGS_PROFILE:STRING=-pg -g3 -ggdb -fno-inline -ftest-coverage -fprofile-arcs -DNDEBUG
[... 2222 lines cut ...]
_PC_TINYXML_LIBS_PATHS:INTERNAL=
_PC_TINYXML_PREFIX:INTERNAL=/usr
_PC_TINYXML_STATIC_CFLAGS:INTERNAL=
_PC_TINYXML_STATIC_CFLAGS_I:INTERNAL=
_PC_TINYXML_STATIC_CFLAGS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_INCLUDE_DIRS:INTERNAL=
_PC_TINYXML_STATIC_LDFLAGS:INTERNAL=-ltinyxml
_PC_TINYXML_STATIC_LDFLAGS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_LIBDIR:INTERNAL=
_PC_TINYXML_STATIC_LIBRARIES:INTERNAL=tinyxml
_PC_TINYXML_STATIC_LIBRARY_DIRS:INTERNAL=
_PC_TINYXML_STATIC_LIBS:INTERNAL=
_PC_TINYXML_STATIC_LIBS_L:INTERNAL=
_PC_TINYXML_STATIC_LIBS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_LIBS_PATHS:INTERNAL=
_PC_TINYXML_VERSION:INTERNAL=2.6.2
_PC_TINYXML_tinyxml_INCLUDEDIR:INTERNAL=
_PC_TINYXML_tinyxml_LIBDIR:INTERNAL=
_PC_TINYXML_tinyxml_PREFIX:INTERNAL=
_PC_TINYXML_tinyxml_VERSION:INTERNAL=
__pkg_config_checked__PC_ACE:INTERNAL=1
__pkg_config_checked__PC_GTKDATABOX:INTERNAL=1
__pkg_config_checked__PC_GTKDATABOXMM:INTERNAL=1
__pkg_config_checked__PC_SQLITE:INTERNAL=1
__pkg_config_checked__PC_TINYXML:INTERNAL=1

drdanz, 2015-08-27:
This is a known issue. See #368.
Unfortunately at the moment it is not possible to use `ALLOW_IDL_GENERATION` on the first build, so you can disable the flag, build, enable the flag, build again.

I started working on this some time ago (see #376) but I never managed to finish that patch... Maybe sometimes in the future...

Anyway, closing since it's a duplicate of #368

closed 2015-08-27 by drdanz