robotology/yarp #564: yarp clock crash on useSystemClock() call

barbalberto, 2015-08-17:
When coming back ***from networkClock to systemClock*** I get a segfault in NetworkClock.cpp line 101, function NetworkClock::delay @ waiters.erase(waiterIterator);

I use a 'yarp write /clock' to send the timing. After sending '5 0' the main wakes up and then I get the segfault (not all the times but 90%)

Test code hereafter:
``` 
------------------------------------------------------------------
#include <yarp/os/all.h>
#include <yarp/os/RateThread.h>

#include <yarp/dev/all.h>
#include <yarp/sig/all.h>

using namespace std;
using namespace yarp::os;
using namespace yarp::dev;
using namespace yarp::sig;

/***********************************************/
class ThreadTest : public yarp::os::RateThread
{
    public:
        ThreadTest() : RateThread(10) { }

        bool threadInit() {  yarp::os::Time::useNetworkClock("/clock"); }
        void run() { }
        void threadRelease() { }
};

/***********************************************/
int main(int argc, char *argv[])
{    
    Network yarp;
    if (!yarp.checkNetwork())
        return -1;

    std::cout << "START" << std::endl;

    yarp::os::Time::useNetworkClock("/clock");
    ThreadTest test1, test2;
    test1.start();
    test2.start();

    yarp::os::Time::delay(5);
    yarp::os::Time::useSystemClock();
    test1.stop();
    test2.stop();
    return 0;
}
```

traversaro, 2015-09-02:
Fix provided in https://github.com/robotology/yarp/pull/575/files . 
Basically the `useSystemClock()` (through the NetworkClock destructor) was waking the `RateThread` slave threads, that right after waking they were calling `yarp::os::Time::now()`. Unfortunately the `useSystemClock()` in the parent thread was still not finished when `yarp::os::Time::now()` was called in the child link, and so the `now()` method of NetworkClock was called, even if the NetworkClock was being destroyed! 
Properly fixing the issue would require a mutex in `yarp::os::Time::now()`, that is not desirable in non-simulation scenario. The alternative solution provided in  https://github.com/robotology/yarp/pull/575 is to lock the mutex only if `yarp::os::Time` is using an external clock... in this way the overhead for the non-simulation scenario is just a check that the `pclock` is null.

traversaro, 2015-09-02:
Just to be clear: this PR does not solve all synchronization issues in the use of NetworkClock , but I think is a nice and simple tradeoff to solve this bug.

traversaro, 2015-09-02:
Actually there is a better method for fixing this, I will implement itas soon as I have some time.

barbalberto, 2015-10-13:
Can you explain your idea about how to fix this? Maybe I can implement it

traversaro, 2015-10-13:
It was a Fermat-like statement.  : ) 
Clearly I don't remember anything about this issue, but I recently found the solution that I had implemented that night and pushed it in https://github.com/robotology/yarp/commit/f734a4c01961def45cdb066c1192446a83088fb0 . The thing that I remember that in the end I was not fully satisfied even by this solution because it was still possible to get race conditions even with this solution, but I don't remember the details.

pattacini, 2015-10-13:
:crown: on @traversaro for the citation :smiley:

traversaro, 2015-10-13:
:D 
One last thing: both fixes can still (theoretically) lead to some race conditions, but both of them solved the issue for all the local tests that I did.

traversaro, 2015-10-24:
Closed by https://github.com/robotology/yarp/pull/609 .

closed 2015-10-24 by traversaro