robotology/yarp #565: ResourceFinder and nested configuration files

alecive, 2015-08-18:
By following this link: http://wiki.icub.org/yarpdoc/yarp_config_files.html#yarp_config_file_nesting , I have created a `parent.ini` file in such a way that it includes another one with this command:
```
[include mygroup "child.ini"]
```
Everything works great, except the fact that apparently `child.ini` is successfully found only if it is in the same folder in which `parent.ini` is located. In many cases, this might not be the case: for example, the user might change a parameter in `parent.ini` and put it into `.local/share/yarp/..`, whilst keeping `child.ini` with its default parameters into e.g. the install directory. The dual might also be possible (i.e. `child.ini` into `.local` and `parent.ini` into the install directory).

Solution would be to force the `ResourceFinder` to look again into the usual directory structure when a nested file is requested.

Cc @super-ste @julijenv

barbalberto, 2015-08-19:
Other related bug with resourceFinder and includes is #459

lornat75, 2015-08-21:
I would not call this a bug; the include keyword is supposed to work locally with respect to the file itself.

traversaro, 2015-09-02:
A possible option (still to be discussed) is to don't change the behaviour of the existing `include` directive, and include a new directive to mean "include by searching with ResouceFinder".

alecive, 2015-09-02:
@traversaro this indeed could be a nice solution to the problem.

lornat75, 2015-09-02:
I checked this issue, it it not easy to solve :)

The way the RF works now is that it searches for the file using the various policies, then it opens the file with a yarp Property. The yarp Property handles the "include" directive knowing nothing about the RF policies. 

A possible solution could be to add code that:
 1. opens the file
 * parses it to search for include directives
 * extracts the corresponding file names and passes them to the RF to get full paths
 * replaces these filenames with full specified paths
 * passes the new file to the Property as before

It would not handle further nesting of include, though.

randaz81, 2021-01-06:
Sometimes they come back... :-)
In my use case I need to **include a .ini file belonging to a different context**

pattacini, 2021-01-06:
You might have already found a way through it @randaz81, but it is possible to load files belonging to different contexts although this involves a bit of code.

An example of that is provided by `iKinGazeCtrl`, whose [config file][1] foresees to load camera calibration parameters from a different context, actually. In essence, an additional ResourceFinder needs to be instantiated internally with the options retrieved from the main conf file.

[1]: https://github.com/robotology/icub-main/blob/master/app/iKinGazeCtrl/conf/config_imu_filter.ini

randaz81, 2021-01-06:
Hi @pattacini, thank you! I already used the "trick" you described  in the past, I think in the navigation stuff.
However, today, before going with it for the umpteenth time, I tried to check if there was a more elegant solution to the problem... and I finished here :-)
Anyway, in the end (and here I have a surprise... 🥳) I gave a look to the RF class and I implemented the feature (ta-daaan, it also supports multi-level nesting!)
The implementation is a bit rough, though, so I'm going to discuss a couple of things with @drdanz before proceeding further. Anyway, I'd move the technical discussion from here to the PR and I'll put the link here asap.
Happy 2021!

alecive, 2021-01-06:
This is a very good demonstration that I am ahead of my times. Sometime in 2050 somebody will pull up an issue I opened and they would just say: "WOW! This guy"

randaz81, 2021-01-09:
fixed by https://github.com/robotology/yarp/pull/2451

randaz81, 2021-01-09:
Unfortunately, sometimes we need to close old issues, even if we are attached to them 😄 

This issue is now fixed by https://github.com/robotology/yarp/pull/2451. Bye!

closed 2021-01-09 by randaz81