robotology/yarp #566: Matlab Binding Problem after Yarp update

hjchang, 2015-08-19:
Hi. After I recently updated the yarp, I cannot use the matlab binding anymore. It says that the 'LoadYarp' is a undefined function or variable. It was fine before the update. I cannot find any document related to this. Please help me with this problem.

barbalberto, 2015-08-20:
Just a shoot in the dark, try to go in YARP_DIR/build/bindings to look for any matlab related file, remove it and recompile.
It helped me in the past with python bindings, I hope it helps you too

Tobias-Fischer, 2015-08-25:
Hi,
we've just checked this issue again. We deleted the `build` directory, and started from scratch. We solved the problem by doing the following:
1. run `ccmake`, and trigger the `YARP_COMPILE_BINDINGS` and `CREATE_JAVA` and `PREPARE_CLASS_FILES`
2. run `make`
3. `cd build/yarp; cp LoadYarp.class YarpImageHelper.class ..`
4. change `classpath.txt` to contain `$YARP_ROOT/build` (with $YARP_ROOT substituted with the actual path)
5. change `librarypath.txt` to contain `$YARP_ROOT/build/lib` (with $YARP_ROOT substituted with the actual path)
6. run matlab, and call `LoadYarp`

I guess that this procedure could be simplified if `LoadYarp.class` and `YarpImageHelper.class` were copied in the correct folder (in a parent folder of all the other *.class files) by cmake, and thus the CMakeLists.txt would be needed to be modified. I saw that @fstramandinoli was playing with the CMakeLists.txt of the bindings recently, do you know an easy way to accomplish that?

If not, could someone please update the documentation on `http://wiki.icub.org/yarpdoc/yarp_swig.html` (maybe @pattacini ;)?)

hjchang, 2015-08-25:
Hi Tobi,

That's great! Thank you so much:)

Cheers,

Hyung Jin

*Hyung Jin Chang*,  Ph.D.
Research Associate in Personal Robotics Lab.
Department of Electrical and Electronic Engineering
Imperial College London
South Kensington, London, UK
hyungjinchang.wordpress.com

lornat75, 2015-08-26:
I know that @drdanz revised the procedure this summer at the summer school. Daniele can you check this?

drdanz, 2015-08-27:
@fstramandinoli and I made some changes to put all the class files in the right place, but to be honest I don't remember much... from what I understand we created on purpose all the class files in <build>/yarp, but maybe that's not the right place? I will have to check that from Francesca as soon as she is back.

traversaro, 2015-09-02:
This should be fixed by https://github.com/robotology/yarp/pull/572?w=1 , will check with @fstramandinoli if is correctly working on Windows.

closed 2015-09-02 by traversaro