robotology/yarp #569: Improve Yarp capability to write/read image files

pattacini, 2015-08-28:
##### Preamble
`Yarp` has several `write()` and `read()` routines to save and load image files.

At present we can write/read images of the following formats:
- 8-bit (int) monochromatic images
- 32-bit (float) monochromatic images
- `RGB`/`BGR`/`RGBA` (int) color images

For each of the above image types `Yarp` exposes the corresponding API:
```cpp
bool file::write(const ImageOf<PixelMono> &src, const ConstString &dest);
bool file::write(const ImageOf<PixelFloat> &src, const ConstString &dest);
bool file::write(const ImageOf<PixelRgba> &src, const ConstString &dest);
bool file::write(const ImageOf<PixelRgb> &src, const ConstString &dest);
bool file::write(const ImageOf<PixelBgr> &src, const ConstString &dest);
bool file::write(const Image& src, const ConstString& dest);

bool file::read(ImageOf<PixelMono> &dest, const ConstString &src);
bool file::read(ImageOf<PixelFloat> &dest, const ConstString &src);
bool file::read(ImageOf<PixelRgba> &dest, const ConstString &src);
bool file::read(ImageOf<PixelRgb> &dest, const ConstString &src);
bool file::read(ImageOf<PixelBgr> &dest, const ConstString &src);
```

Interestingly, we have only the _write_ part for the generic `yarp::sig::Image` type, lacking of the _read_ counterpart. However, inspecting the body of the latter function we can see that, sadly, an `RGB` image gets always saved.

```cpp
bool file::write(const Image& src, const ConstString& dest) {
    ImageOf<PixelRgb> img;
    img.copy(src);
    return write(img,dest);
}
```

The patch in https://github.com/robotology/yarp/commit/d76fa3ec28f0307ada530f9a53612a7e577e2915 attempts to expand such a behavior into:
```cpp
bool file::write(const Image& src, const ConstString& dest) {
    int code=src.getPixelCode();
    if ((code==VOCAB_PIXEL_MONO) || (code==VOCAB_PIXEL_MONO16))
        return write(static_cast<const ImageOf<PixelMono>&>(src),dest);
    else if (code==VOCAB_PIXEL_MONO_FLOAT)
        return write(static_cast<const ImageOf<PixelFloat>&>(src),dest);
    else if (code==VOCAB_PIXEL_BGR)
        return write(static_cast<const ImageOf<PixelBgr>&>(src),dest);
    else if (code==VOCAB_PIXEL_RGB)
        return write(static_cast<const ImageOf<PixelRgb>&>(src),dest);
    else if (code==VOCAB_PIXEL_RGBA)
        return write(static_cast<const ImageOf<PixelRgba>&>(src),dest);
    else {
        ImageOf<PixelRgb> img;
        img.copy(src);
        return write(img,dest);
    }
}
```

##### Bunch of comments
1. We should better exploit the object hierarchy of `yarp::sig::Image` to achieve the same goal in a much more standard OOP approach, though the design seems quite messy.
2. We still don't have the chance to store the remaining image types.
3. We definitely need the `file::read(Image&,...)` counterpart, otherwise `yarpdatadumper` is able to save generic images while `yarpdataplayer` remains unable to load a generic image from a file.

Note that upon the increasing use of stereo vision, users will be likely requiring to store the exact format of images of types `int_mono` and `float_rgb`.

##### Final Question
Will `yarpdataplayer` be really able to read generic image and correctly populate an `yarp::sig::Image` object?

It seems that the files stored by means of `file::write()` APIs do not contain information on their actual types (needs for double check). That's quite bad since it'd force us to introduce such information alongside with the logged data.

Finally, a quick glance at the routine devoted to save a float image reveals that the format appears to be quite non standard.

/cc @vtikha

francesco-romano, 2015-08-31:
Just to add a side note:
At the present state yarp does not compile without opencv (see #570)

pattacini, 2015-08-31:
@francesco-romano does not compile on the evil Mac ... to be precise :smirk:

pattacini, 2015-08-31:
@francesco-romano donno, I've just merely noted the title of the issue :smile: 
Thus, if the issue is os-independent, please update it accordingly.

vtikha, 2015-08-31:
I confirm that this is not only a osx issue.

pattacini, 2015-09-02:
The idea is to let `yarpdatadumper` store the type of the image too, so that `yarpdataplayer` is able to retrieve such info independently.

vtikha, 2015-09-02:
yarpdataplayer is now able to read the type of the image (recent change in yarpdatadumper) and send it accordingly. This does not break datasets that do not include this new feature.

closed 2015-09-02 by vtikha

pattacini, 2015-09-02:
Following up in #574