robotology/yarp #570: YARP GUI compilation is broken in OS X without OpenCV

traversaro, 2015-08-31:
~~~
mkdir build_no_opencv
ccmake -DYARP_USE_OPENCV:BOOL=FALSE -DCREATE_GUIS:BOOL=TRUE ..
make
...
[ 89%] Building CXX object src/yarpdataplayer-qt/CMakeFiles/yarpdataplayer-qt.dir/src/worker.cpp.o
In file included from /Users/traversaro/src/robotology/yarp/src/yarpdataplayer-qt/src/worker.cpp:26:
/Users/traversaro/src/robotology/yarp/src/yarpdataplayer-qt/include/mainwindow.h:74:10: warning: 'MainWindow::attach' hides overloaded virtual functions [-Woverloaded-virtual]
    bool attach(yarp::os::RpcServer &source);
         ^
/Users/traversaro/src/robotology/yarp/src/libYARP_OS/include/yarp/os/Module.h:160:18: note: hidden overloaded virtual function 'yarp::os::Module::attach' declared here: type mismatch at 1st parameter
      ('yarp::os::Port &' vs 'yarp::os::RpcServer &')
    virtual bool attach(Port& port);
                 ^
/Users/traversaro/src/robotology/yarp/src/libYARP_OS/include/yarp/os/Module.h:171:18: note: hidden overloaded virtual function 'yarp::os::Module::attach' declared here: different number of parameters (2 vs 1)
    virtual bool attach(TypedReader<Bottle>& port, bool handleStream = false);
                 ^
/Users/traversaro/src/robotology/yarp/src/yarpdataplayer-qt/src/worker.cpp:162:11: error: no matching function for call to 'read'
    if ( !read(img,tmpPath.c_str()) ) {
          ^~~~
/Users/traversaro/src/robotology/yarp/src/libYARP_sig/include/yarp/sig/ImageFile.h:42:31: note: candidate function not viable: cannot bind base class object of type 'yarp::sig::Image' to derived class reference
      'ImageOf<yarp::sig::PixelRgb> &' for 1st argument
            bool YARP_sig_API read(ImageOf<PixelRgb>& dest, 
                              ^
/Users/traversaro/src/robotology/yarp/src/libYARP_sig/include/yarp/sig/ImageFile.h:44:31: note: candidate function not viable: cannot bind base class object of type 'yarp::sig::Image' to derived class reference
      'ImageOf<yarp::sig::PixelBgr> &' for 1st argument
            bool YARP_sig_API read(ImageOf<PixelBgr>& dest, 
                              ^
/Users/traversaro/src/robotology/yarp/src/libYARP_sig/include/yarp/sig/ImageFile.h:46:31: note: candidate function not viable: cannot bind base class object of type 'yarp::sig::Image' to derived class reference
      'ImageOf<yarp::sig::PixelRgba> &' for 1st argument
            bool YARP_sig_API read(ImageOf<PixelRgba>& dest, 
                              ^
/Users/traversaro/src/robotology/yarp/src/libYARP_sig/include/yarp/sig/ImageFile.h:48:31: note: candidate function not viable: cannot bind base class object of type 'yarp::sig::Image' to derived class reference
      'ImageOf<PixelMono> &' for 1st argument
            bool YARP_sig_API read(ImageOf<PixelMono>& dest, 
                              ^
/Users/traversaro/src/robotology/yarp/src/libYARP_sig/include/yarp/sig/ImageFile.h:52:31: note: candidate function not viable: cannot bind base class object of type 'yarp::sig::Image' to derived class reference
      'ImageOf<PixelFloat> &' for 1st argument
            bool YARP_sig_API read(ImageOf<PixelFloat>& dest, 
                              ^
1 warning and 1 error generated.
make[2]: *** [src/yarpdataplayer-qt/CMakeFiles/yarpdataplayer-qt.dir/src/worker.cpp.o] Error 1
make[1]: *** [src/yarpdataplayer-qt/CMakeFiles/yarpdataplayer-qt.dir/all] Error 2
make: *** [all] Error 2

~~~

traversaro, 2015-08-31:
OS X Yosemite, yarp from master, clang: 
~~~
clang -v
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
~~~

francesco-romano, 2015-08-31:
see https://github.com/robotology/yarp/issues/569

pattacini, 2015-08-31:
I think @vtikha might help on this.

francesco-romano, 2015-08-31:
As a temporary workaround @vtikha suggested to put back the `ImageOf<PixelRgb>` instead of `Image`

pattacini, 2015-08-31:
@francesco-romano please go ahead with the workaround, thanks.

closed 2015-08-31 by francesco-romano

vtikha, 2015-08-31:
Indeed this is related to #569. In YARP we have only the `write` part for the generic `yarp::sig::Image` type, and no read equivalent. However, since the generic Image is saving in RGB we can go ahead with the workaround `yarp::sig::ImageOf<yarp::sig::PixelRgb>` and improve yarp's routines without opencv in order to cope with recent developments eg: the stereo-vision which requires to store the format of images of types `int_mono` and `float_rgb`.