robotology/yarp #576: Get rid of yarp::os::Module

pattacini, 2015-09-02:
`yarp::os::Module` class is practically deprecated in favor of `yarp::os::RFModule`, hence we should consider removing it.

Before removing, we need to update `yarpdataplayer-qt` to the use of `yarp::os::RFModule` (/cc @vtikha).

Fix https://github.com/robotology/icub-main/issues/191 as a pre-requisite.
See also #432.

/cc @alecive

vtikha, 2015-09-04:
This is not actually a module per se...cleaned it up

pattacini, 2015-09-09:
Actually, this ticket was concerned with the idea of drastically removing the class yarp::os::Module from Yarp. Thus, reopening for discussion.

vtikha, 2015-09-09:
I am removing the `yarpdataplayer-qt` tag which is erroneous in this case.

traversaro, 2015-09-09:
I think we should start to use the `YARP_DEPRECATED` introduced some time ago that is now disable by default, so we can induce warnings on 3rd party code that uses classes/function that we want to deprecated. `yarp::os::Module` seems to be a good use case for this.

drdanz, 2015-09-09:
+1 for marking it ``YARP_DEPRECATED``. I would also remove it from ``yarp/os/all.h`` and a deprecated warning when including the ``yarp/os/Module.h`` header.
Actually yesterday I tried to do it, but I got stuck with a stupid [issue](https://stackoverflow.com/questions/32463839/inheritance-from-deprecated-class) with gcc.

Also, even if we don't support it, if it is not a long task, I think we should get rid of it in yarpdataplayer-gtk, so that we can theoretically remove the class any time without breaking YARP itself.

lornat75, 2015-09-09:
Fine with me although I would consider it a minor issue.

pattacini, 2015-09-12:
We should be now safe: we no longer make use of `yarp::os::Module` within all the projects we manage (`icub-main`, `icub-basic-demos`, `iol` ...).

@drdanz are you willing to declare `yarp::os::Module` deprecated in the proper way you guys mentioned?

drdanz, 2015-09-12:
Sure! Assigning it to myself.

commit d7f67c8b (2015-11-27): [YARP_OS] Deprecate yarp::os::Module

Fixes #576
closed 2015-11-27 by drdanz