robotology/yarp #578: RFModule online documentation polluted

gsaponaro, 2015-09-03:
Hi, in the online documentation at http://wiki.icub.org/yarpdoc/classyarp_1_1os_1_1RFModule.html I noticed this line under a few functions, it probably should not be there, right?

[image]

pattacini, 2015-09-03:
It's normal, since we have this [**`KinectThread`**](https://github.com/robotology/yarp/blob/master/src%2Fmodules%2Fkinect%2FOpenNI%2FTEST%2Fmain.cpp#L30) class in yarp.

Then, why there's an inheritance from `yarp::os::Thread` and `yarp::os::RFmodule` is at least questionable (`yarp::os::RFModule` is a thread in the end).

closed 2015-09-03 by pattacini

gsaponaro, 2015-09-03:
I see, thanks.