robotology/yarp #580: [OSX] Full OS X support

francesco-romano, 2015-09-07:
This issue summerizes the currently open issues to have a full OS X support for yarp applications.

- [ ] Application Icons
- [ ] Menu or various icons (it seems they should be .png. We must check this requirements)
- [ ] Environmental variables (we should find alternatives, e.g. `YARP_DATA_DIRS` vs `path.d`
- [ ] Relocation of `.app` at the current state does not work
- [ ] Optional generation of `.framework` bundle
- [ ] GUI should allow an alternative to (most) command line parameters (e.g. yarpview port name)

cc @vtikha @drdanz @lornat75

alecive, 2015-09-07:
Related to #1: if you give me sane ideas for the app icons, I'm willing to do them
Related to #2: on't know about mac OSX, but qt supports `.svg` icons (this is what I started using for the guis - it scales even for you retina-crazy lovers).

alecive, 2015-09-07:
e.g. this was a "sane" idea for the `iCubWorld` icon (from @ccliber ): https://github.com/alecive/iCubArtwork/blob/master/iCub-Icon/iCub_World_solidcolor.svg

vtikha, 2015-09-07:
Related to #1: @alecive thanks, let's quickly meet and see what we come up with
Related to #2: @alecive is right, I just checked it all and `.svg` is good aslo for OSX

francesco-romano, 2015-09-07:
Then we have to understand why all menus are without icons...

kt10aan, 2015-09-07:
Well, the standard menu style in OS X does not use icons, so maybe there is an option to bring the menu items to be aligned to all the way to the left?

Regarding the last point, the availability of at least the basic parameters from the GUI would be also nice for the linux apps.

francesco-romano, 2015-09-07:
@alecive this can be useful: https://developer.apple.com/library/mac/documentation/UserExperience/Conceptual/OSXHIGuidelines/Designing.html

But I think a generic icon for OS X and other system will be great anyway :)

francesco-romano, 2015-09-07:
@kt10aan you're right. Even if icons are supported (see for example the history of safari)

alecive, 2015-09-07:
@francesco-romano don't know if you're kidding or serious : I do know how to design icons (whilst OSX designers don't) :smile:

kt10aan, 2015-09-09:
Regarding the bundle generation (which is optional now as I understand): I can update the Homebrew formula to have an extra option for bundle generation. Also, the formula could optionally link the generated apps in the Applications folder (while the actual apps remain in ``/usr/local/Cellar/yarp/...``

francesco-romano, 2015-09-09:
Adding @drdanz comments at the (now merged) PR:

~~~
For some GUIs yarp_osx_bundle_update_rpath (and some other commands) are called only if(YARP_OSX_GENERATE_BUNDLES), for other are always called. Is this the wanted behaviour?
The other icons are all in /data/icons, I think these .icns icons should be there as well.
~~~

francesco-romano, 2015-09-09:
@drdanz 
1. No. Simply because I forgot to move that part in the if. It should not affect the behaviour anyway. I added it to the task list.
2. I though that generic icons were in `data/icons`. Because these are application specific I put them here.

francesco-romano, 2015-09-19:
After a (short) trial I think we cannot generate the frameworks.
The issue lies in how the headers are packed in the framework.
For example, AppKit:

[image]

And the headers are included with `#include <FrameworkName/HeaderName.h>`, so `#include <AppKit/AppKit.h>`.

But in yarp we don't have a flat hierarchy for the headers, for example `yarp/os/...`. 
So even if the framework are easily generated, the current code use them. A proper use can be
`#include <YARP_OS/os.h>`

@drdanz

francesco-romano, 2015-09-29:
These can help...
http://cmake.org/cmake/help/git-master/module/BundleUtilities.html

Thanks @drdanz for the link.

kt10aan, 2015-09-29:
Ah yes:). Also, let's see what changes 10.11 will bring!

francesco-romano, 2015-10-12:
Added another probable issue: how `.app` integrates with `yarpmanager`?

@apaikan

francesco-romano, 2015-10-12:
Fast poll: 
As the current state `.app` support is not ready to substitute the command line utilities.
The question is: should we change the option to generate the bundles to generate ALSO the bundles, meaning that the command line utilities will be always generated?

@vtikha @traversaro @drdanz @kt10aan @iron76 @ChiaraBartolozzi

kt10aan, 2015-10-12:
My opinion is that both command line and apps should be generated. Not only because the apps are not ready yet,but because sometimes the command line utilities are useful for debugging. Also, in case of any regression on the future, command line versions of the app will be available as backup :)

In any case, when the cmake options regarding the apps are finalized, I will add the ''brew link apps'' option to the formula.

vtikha, 2015-10-12:
In theory, it would be cleaner to generate things depending upon user's choice e.g. just command line, just bundles or both...but, I do agree with @kt10aan and until we fix things properly I will also suggest to generate both.  
:+1: @kt10aan for the formula

ChiaraBartolozzi, 2015-10-12:
I agree with @kt10aan, otherwise, if you want to let the user choose, then make sure the user does know about the option :-)

btw will the .app work with yarpmanager?

claudiofantacci, 2017-03-08:
We should really sweep up and update this issue.
Let's procede in order:

- **Application Icons**: it seems to me that we have icons, but some are still missing. Am I right?
- **Menu or various icons**: are me really missing icons in menu?
- **Environmental variables**: ???
- **Relocation of `.app`** + **GUI should allow an alternative to (most) command line parameters**: I think that these two features can be tackled in one shot. Note that a bundled app is nothing but a "folder" with the compiled executable and some other files (like the icon, plist, etc.). We can make full advantage of a bundles by replacing the standard CL executables (which are also present in the bundle app, hence redundant) with symbolic links to the executables in the apps so that we can have both the advantages of having an app (with icons and other options) and the features provided by the CLI (like passing specific parameters at runtime).
- **Menu shortcut**: I think they are still missing.
- **`yarpmanager` integration**: what kind of integration are we want to achieve?

francesco-romano, 2017-03-08:
@claudiofantacci  Answering to your points:

- **Application Icons**: yes. I think some apps are missing the icon. Furthermore their quality is not correct. I don't know the resolution (which should be at least 1024x1024 pixels “see here](https://developer.apple.com/library/content/documentation/UserExperience/Conceptual/OSXHIGuidelines/Designing.html), but they are missing transparency (e.g. yarpview)
- **Menu or various icons**: probably solved in the meantime. I think it was an issue of Qt5.
- **Environmental variables**: at least two behaviours in YARP are controlled by environment variables: `YARP_DATA_DIRS` and `YARP_ROBOT_NAME`. When an application is launched in macOS it is not a  child of bash thus it does not inherit the environment variables set in the `bash_profile` file. To be GUI-friendly we should find other ways of setting those properties.
- **Relocation of .app + GUI should allow an alternative to (most) command line parameters**: I don't think they are the same issue. GUI alternative means that all the parameters that at this stage are passed **ONLY** as command line parameters should be configurable in some way also from menu actions. Relocation instead means being able to move the `.app` anywhere on the disk. And now I remember why I did not create the symbolic link. That would completely break the relocation feature. 
But because without a complete rewriting of the YARP structure relocation is difficult, we can implement the symbolic link feature and forget about relocation 
- **Menu shortcut**: as you said
- **yarpmanager integration**: I don't remember 😅

claudiofantacci, 2017-03-08:
@francesco-romano ok on everything.

I think we can start with the symbolic link feature and then try to move on the others.

I also suggest to update the first post as a follow up of this discussion.