robotology/yarp #582: yarpdev error when starting cameras: feature not present

Tobias-Fischer, 2015-09-08:
Hi,
when starting up our dragonfly cameras on the iCub with `yarpdev --from camera/dragonfly2_config_left_320_240.ini`, we get a bunch of errors:
```
3472		73.318079		DEBUG		Format = 1
3474		73.358501				yarp: Port /log/pc104/yarpdev/3173 active at tcp://10.0.0.2:10801
3475		73.368613				yarp: Sending output from /log/pc104/yarpdev/3173 to /yarplogger using tcp
3477		73.390946				------ Camera information ------
3479		73.391071				Vendor                            :     Point Grey Research
3480		73.391112				Model                             :     Dragonfly2 DR2-03S2C-EX
3481		73.391152				Unit                              :     0
3482		73.391189				Specifications ID                 :     0xa02d
3483		73.391226				Software revision                 :     0x102
3484		73.391264				IIDC version code                 :     548
3485		73.391299				Unit directory offset             :     0x428
3487		73.401026				Unit dependent directory offset   :     0x440
3488		73.401087				Commands registers base           :     0xf00000
3489		73.401128				Unique ID                         :     0x00b09d01009a72ed
3490		73.401230				Vendor ID                         :     0xb09d
3492		73.401346				Model ID                          :     0x1
3493		73.401383				Advanced features found at offset :     0xf01000
3494		73.401452				1394b mode capable (>=800Mbit/s)  :     Yes
3495		73.401534				Platform backend                  :     juju
3496		73.401582				------ Camera platform-specific information ------
3497		73.401620				Device filename                   :     /dev/fw1
3498		73.401656				
3499		73.401691				fps=25 newBandOcc=13593600.000000 bpp=3.000000 bytesPerPacket=1056 maxBytesPerPacket=1920
3500		73.401730				
3501		73.401764		ERROR		Feature 426 not present
3503		73.401841		ERROR		Feature 430 not present
3504		73.401931		ERROR		Feature 432 not present
3506		73.411122		ERROR		Feature 435 not present
3509		73.412277		ERROR		Feature 436 not present
3511		73.412346		ERROR		Feature 437 not present
3514		73.421234		INFO		created device <dragonfly2>. See C++ class yarp::dev::DragonflyDeviceDriver2Rgb for documentation.
3552		73.481784				yarp: Port /icub/cam/right active at tcp://10.0.0.2:10241
3553		73.481844				yarp: Setting framerate to: 25[Hz] (thread period 40[ms])
3554		73.481903				
3555		73.481940		INFO		created wrapper <grabber>. See C++ class yarp::dev::ServerFrameGrabber for documentation.
3562		74.208614				yarp: Port /icub/cam/right/quit active at tcp://10.0.0.2:10243
3564		74.208706		INFO		device active...
```

However, the errors do not affect the functionality of the cameras. Everything is working fine. It would be great if someone has a look what those errors mean, and whether they are really errors or more like warnings / infos.

Thanks a lot and best wishes,
Tobias

PS: I guess this issue has minor priority.

barbalberto, 2015-11-20:
If I remember correctly, when the camera device starts, it ask the real camera for all possible settings, like saturation, brightness and so on ( to configure the buttons on the gui, for example).
This message simply tell that the requested feature is not implemented by the specific HW. Different cameras implement different subset of feature. So this is not an error that will compromize the functionality, simply will disable the relevant button on the gui.
I guess the message is misleding and should be changed from 'error' to 'debug' or similar.

Tobias-Fischer, 2016-08-01:
Hi,
This issue has been open for a while now, and the upcoming IFI day might be a good opportunity for a fix. I agree that using a debug message is more appropriate, and would be happy to create a PR if other people see this as a suitable solution as well :).
Best, Tobi

barbalberto, 2016-10-06:
This should be fixed by commit #4f33aa1d186438fa36e77e1486be64ff7c2a6691.
Can we close this issue now?

closed 2016-11-03 by barbalberto