robotology/yarp #584: calibrate button not enabled

valegagge, 2015-09-10:
calibrate button is not enabled when control mode is not_configured, even if yarpmotogui is launched with --calib param.

valegagge, 2015-10-09:
same problem even if yarpmotogui is launched with --admin param.

randaz81, 2015-10-09:
I thought it was fixed by e159a1fc36fce1b380ee13b80811c3ab23ff785c, jointitem.cpp. I'll check why it's still not working

davege89, 2015-10-14:
Today I gave a look through the code of the GUI.
I think I've found the cause of the problem:
when the state of the joint is *NotConfigured*, the *SetJointState* method of the **JointItem** object disables the buttonsContainer, containing the possible commands for the single joint (Idle, Home, Calib, Run, Pid). This makes the enabling of the *calib* button completely useless. 
I tried to change this part:
```c
...
ui->buttonsContainer->setEnabled(false);
if (enableCalib){
   ui->buttonCalib->setEnabled(true);
}
...
```
with this one:
```c
...
if(enableCalib)
{
   ui->buttonsContainer->setEnabled(true);
   ui->buttonIdle->setEnabled(false);
   ui->buttonPid->setEnabled(false);
   ui->buttonHome->setEnabled(false);
   ui->buttonCalib->setEnabled(true);
   ui->buttonRun->setEnabled(false);
   ui->comboMode->setEnabled(false);
   ui->comboInteraction->setEnabled(false);
 }
 //nothing must be activated
 else
 {
    ui->buttonsContainer->setEnabled(false);
 }
...
```
This version is working on my local machine, letting the user click only the *calib* button for the described situation.
Let me know if this could be a good solution, so that I can commit it in the next few days.

closed 2015-10-15 by davege89

barbalberto, 2015-10-16:
:+1: :+1:

commit 7a59871f (2015-10-15): Fixes #584