robotology/yarp #587: NITE for YARP kinect unavailable

tanismar, 2015-09-17:
I have been trying to compile YARP with OpenNI (and 2) in order to work with Kinect (and other 3D sensors), but in the current version, also NITE2 library is required for YARP to compile properly. The problem is that since Apple bought Primesense in 2013, NITE2 is not open anymore and hence unavailable. Moreover, although skeleton recognition is awesome, it would be more useful to have a way for processing depth images on YARP, with the option to include also Skeleton and gesture detection with NITE (not NITE 2, as it is unavailable), but not making it a compulsory dependency. 
Summarizing: Is it possible to modify the OpenNI2 device driver for YARP so that it does not require NITE2? 
Thanks in advance!
Tanis

drdanz, 2015-09-17:
@kt10aan Can you have a look at this issue, please?

kt10aan, 2015-09-17:
OpenNI1 works NiTE1 and OpenNI2 with NiTE2. You cannot mix them.
OpenNI1 Is very old now and unsupported with some nasty bugs. OpenNI2 is actively maintained.

Making NiTE2 an optional dependency is for some time now in my todo list, but I just have to find the time to do it! 

By the way, NiTE2 is still available with a bit of googling. If you are in a hurry, send me a pm or email and I will send you the link :)

I will also add to the documentation the link to the NiTE2 binaries tomorrow morning :)

tanismar, 2015-09-17:
Hi Kostas, thanks for the quick response! I did quite some googling, but I couldnt find NITE2, all lead to Apple products websites. So in any case, if you can send me the link (or the code) for a fast hack would be nice, but I think that dropping the dependency would be better as a long term solution :). 
Best!
Tanis

kt10aan, 2015-09-17:
I think you already got the link :)
Yes, decoupling the NiTE2 dependency will happen :)

closed 2015-09-17 by kt10aan