robotology/yarp #588: Windows installer for YARP does not update PATH environment variable

lornat75, 2015-09-22:
System: Visual Studio 2012, installer: 
https://github.com/robotology/yarp/releases/download/v2.3.64/yarp_2.3.64_v12_x86_1.exe

PATH should be updated to incude:
ACE_ROOT/bin
Qt5_DIR/../../../bin

@mbrunettini

lornat75, 2015-09-22:
We could not replicate this problem, closing for now.

closed 2015-09-22 by lornat75

FrankOcean11, 2015-09-22:
which Yarp program should i download for my home built Humanoid ? and from where? [link?] for win8.1

pattacini, 2015-09-22:
Hi @FrankOcean11
Yarp can be installed as binary package or compiled from sources; instructions for Windows are available here: http://wiki.icub.org/wiki/ICub_Software_Installation#Installation_on_Windows.

Btw, official Yarp documentation can be found at this link: http://wiki.icub.org/yarpdoc/index.html.

Should you have further questions/doubts, please consider next time opening a new issue and/or posting on [**`Q&A`**](https://github.com/robotology/QA).