robotology/yarp #589: ROS to Yarp "Sorry, cannot handle [bayer_grbg8] images yet."

WurmD, 2015-09-22:
Using YARP version 2.3.64.6 compiled with rossrv tcpros xmlrpc carriers, ```yarp read /log_bot/kinect/rgb/image_raw@/blahd```

> yarp: Port /blahd active at tcp://10.0.28.138:10003
> yarp: Port /log_bot/kinect/rgb/image_raw-@/blahd active at tcp://10.0.28.138:10002
> yarp: Receiving input from /play_1442925511934563299 to /log_bot/kinect/rgb/image_raw-@/blahd using tcpros+role.pub+topic./log_bot/kinect/rgb/image_raw
> **Sorry, cannot handle [bayer_grbg8] images yet.**

These are images from a Kinect v1 stored in a rosbag.
Is support for this type something we can hope for in the near future?

/cc @plinioMoreno

FrankOcean11, 2015-09-22:
which Yarp program should i download for my home built Humanoid ? and from where? [link?] for win8.1

lornat75, 2015-09-23:
Actually we may need some interoperability between Kinect images and yarp, @drdanz.

lornat75, 2015-10-08:
Sorry for the delay, we can add a case to handle bayer encoding here (thanks @paulfitz):

https://github.com/robotology/yarp/blob/5b0f8a4bfe22ba5bab1696b21438def6c92f21cd/src/carriers/wire_rep_utils/WireTwiddler.cpp#L425

I am a little confused by the meaning of the bayer_grbg8; I guess it means that the image is 1 channel and that pixels have to be interpreted using grbg bayer. In this case we could map this type into a grayscale 8 bits to begin with. If on the other hand the image is 4 bytes for each pixel we have to use another YARP format.

lornat75, 2015-10-12:
@WurmD if you can prepare a test (or just give instruction on how to reproduce it) without a kienct (i.e.using a rosbag) we can try to solve this issue.

WurmD, 2015-10-12:
Sure think @lornat75 

I actually already had one at the other issue :) #591

1)
in terminal 1:
```roscore ```

2)
in terminal 2:
```yarpserver --ros```

3)
in terminal 3:
```rosbag play -l 2015-07-23-16-11-59_0.bag``` https://www.dropbox.com/s/p6rmeemb4h1zj6m/2015-07-23-16-11-59_0.bag?dl=0
which creates node */play_1443117251324113330* and topic */log_bot/kinect/rgb/image_raw*

4)
in terminal 4:
```yarp read /log_bot/kinect/rgb/image_raw@/cuil```
*Sorry, cannot handle [bayer_grbg8] images yet.*

WurmD, 2015-10-12:
My notes investigating the issue:

In http://docs.ros.org/api/sensor_msgs/html/image__encodings_8h_source.html#l00130 numChannels returns 1 when asked for the number of channels of a bayer_grbg8 image.
```
00130     static inline int numChannels(const std::string& encoding)
00146       if (encoding == BAYER_RGGB8 ||
00147           encoding == BAYER_BGGR8 ||
00148           encoding == BAYER_GBRG8 ||
00149           encoding == BAYER_GRBG8 ||
00150           encoding == BAYER_RGGB16 ||
00151           encoding == BAYER_BGGR16 ||
00152           encoding == BAYER_GBRG16 ||
00153           encoding == BAYER_GRBG16)
00154         return 1;
```

In http://docs.ros.org/api/cv_bridge/html/c++/cv__bridge_8cpp_source.html#l00200
```
00052 int getCvType(const std::string& encoding)
00070   if (encoding == enc::BAYER_GRBG8) return CV_8UC1;
````

Nevertheless, in http://docs.ros.org/api/cv_bridge/html/c++/cv__bridge_8cpp_source.html#l00267
```00268 CvImagePtr toCvCopyImpl``` is able to correctly transform bayer_grbg8 into a color RGB8.

So it is not clear what's the inner format of an bayer_grbg8 image.

More things, in http://docs.ros.org/api/cv_bridge/html/c++/cv__bridge_8cpp_source.html#l00182 , the function ```getConversionCodes``` returns a mapping, 
```00132 std::map<std::pair<Encoding, Encoding>, std::vector<int> > getConversionCodes() ```
that is used in ```getConversionCode```
```00188 const std::vector<int> getConversionCode(std::string src_encoding, std::string dst_encoding)``` that returns a specific code for the desired conversion, that is used in
```
00268 CvImagePtr toCvCopyImpl(const cv::Mat& source,
00269                         const std_msgs::Header& src_header,
00270                         const std::string& src_encoding,
00271                         const std::string& dst_encoding)
````
that is actually capable of doing the desired conversion.

lornat75, 2015-10-21:
This commit allows to read images as 8bits single channel. De-bayering needs to be done explicitly.

I think we could extend the code to perform de-bayering and return a color image, however it requires  more work. In addition I am not sure it is a good idea because de-bayering can be done in different ways depending on the needs (NN, interpolation etc) so it could make sense to delegate it to the user.

What do you think?

lornat75, 2015-10-21:
@WurmD: I have implemented some debayering, see  PR #615.

plinioMoreno, 2015-10-21:
In fact the de-bayering requires a lot more work. In the processing suggestion, what kind of pixel should have ImageOf? It should be a PixelMono having an image of 4 times the actual image size? And then the user gets the raw data pointer.

lornat75, 2015-10-22:
The PixelMono image should have the same size as the original image. Bayer encoding places color information in a single byte for each pixel, therefore it perfectly fit in a PixelMono image.

Have a look at the PR, in the end I have implemented the debayering. It is handy but less efficient and accurate of what you could do if you read the image in a PixelMono image and do debayering yourself (with some care of course :) ). It is probably good enough for yarpview.

lornat75, 2015-10-23:
Sorry I messed up #615 so I closed it and made a new PR #620.

drdanz, 2015-11-27:
Should be fixed by #620. Closing this

closed 2015-11-27 by drdanz