robotology/yarp #591: ROS to Yarp, failure to view

WurmD, 2015-09-24:
1)
in terminal 1:
```roscore ```

2)
in terminal 2:
```yarpserver --ros```

3)
in terminal 3:
```rosbag play -l 2015-07-23-16-11-59_0.bag``` https://www.dropbox.com/sh/i98qbj1ua1edsh2/AACK4J9y5q6rLOo9XXzzjut3a?dl=0
which creates node */play_1443117251324113330* and topic */log_bot/kinect/rgb/image_raw*

4)
in terminal 4:
```yarp read /log_bot/kinect/rgb/image_raw@/cuil```
*Sorry, cannot handle [bayer_grbg8] images yet.* That is a separate issue, #589 

My new issue is:
5)
Start python script https://gist.github.com/WurmD/c0641d0277c18c14ebe1 that connects to that topic and converts image to "bgr8" , and outputs to
node */image_converter_16049_1443117432895* and topic */image_topic_2* of this type:
```
header: 
  seq: 18
  stamp: 
    secs: 0
    nsecs: 0
  frame_id: ''
height: 480
width: 640
encoding: bgr8
is_bigendian: 0
step: 1920
```

6)
```yarp read /image_topic_2@/kso```
works, shows pixel values in the terminal

7)
```yarpview --name /view```

8)
if we stop the yarp read and then do
```yarp connect /log_bot/kinect/rgb/image_raw@/cuil /view```
we get:
```Cannot make connection to 'tcp://log_bot/kinect/rgb/image_raw@/cuil'```

9)
if we restart the reader
```yarp read /image_topic_2@/kso```
and do
```yarp connect /image_topic_2@/kso /view```
we obtain
```Success: Added connection from /kso to tcp://view```
but nothing is shown in viewer, black screen, 0 FPS

I'm at a loss :( What else should I try to diagnose this?

YARP version 2.3.64.6
 * /rosdistro: indigo
 * /rosversion: 1.11.13

/cc @plinioMoreno

WurmD, 2015-09-25:
Because yarp connect is not working (steps 8 and 9 above)
we did this code to view the images https://gist.github.com/WurmD/547dc63c34b1b543cb8e

The key lines of code:
```bool outputOk = xd_inputPort.open("/image_topic_2@/my_test");```
```source = xd_inputPort.read(false);```

 - So far the only way to access ROS topics in YARP is to execute ```yarp read <topic name in yarp style>``` or ```yarp write <topic in yarp style>```. Thus, the existing ROS topics are not seen by YARP wihtout executing either a read or write.
 - I guess that since the topics are not visible by yarpserver, the connect cannot find the ports.
 - What would need to be done to have the ROS topics visible by the yarpserver?
 - What's lacking for yarp connect to work?

/cc @drdanz @barbalberto @lornat75

plinioMoreno, 2015-10-06:
The drawback of the behavior explained by @WurmD is that the only way to access a ROS topic in YARP is to write C++ code to read the topic (and from the command line execute the node that reads the topic). It would be very nice if the ROS topics would be visible by the yarpserver, so the yarp connect would work.

What are your thoughts here @barbalberto?

lornat75, 2015-10-08:
@WurmD , @plinioMoreno: this is strange, can you replicate the behavior you describe with other topics that do not involve images?

barbalberto, 2015-10-08:
I run a few tests. It seems to me that using the command
```yarp read /topic@/node```
it works because somehow yarp read detects it is connecting to a ros topic and therefore it does the extra work needed to correctly connect.
Instead when using 
```yarp connect  /topic@/node /yarpPort```
it does not understand it is connecting to a ros topic therefore it is not doing the required extra work.
I also tried specifying the tcpros carrier while connecting but does not solve the issue.
It does not seems to depend on the data type

plinioMoreno, 2015-10-08:
@lornat75 , @barbalberto ,

The same behavior with a simple example:

The [ros node]( https://gist.github.com/plinioMoreno/1e8e1f280fd491fd2936) that publishes "hello world".

The [yarp c++ code](https://gist.github.com/plinioMoreno/b6d0a894c3cf74478a22) that is able to read the ROS topic.

The test:

1. ``roscore``
2. ``yarpserver --ros``
3. ``ros_hello_world_publisher.py``
4. ``yarpidl_rosmsg --name /typ@/yarpidl``

If I run ``yarp read /topic_name@sdfdd`` I am able to read the messages. But I run ``yarp read ... /topic_name@/sdfdd`` I get:
```
yarp: Port /tmp/port/1 active at tcp://10.0.28.138:10012
Failure: could not find source port /topic_name@/sdfdd
```
But using the [yarp c++ code](https://gist.github.com/plinioMoreno/b6d0a894c3cf74478a22) I am able to read the messages. Thus, I can open ROS topics from code and read the messages, but I cannot connect to them using ``yarp connect`` or ``yarp read ... <topic name>``. So the ROS topics are only visible through code. It would be nice if with another option, the ``yarp name list`` would list the ROS topics too.

I wonder if @paulfitz can comment on how would be possible to implement this. The use case of this issue as follows:  An already developed YARP code that creates a port to read data, but we would like to use the same code to connect the YARP code reader to a ROS topic (wihtout hardcoding the code to the ROS topic).

paulfitz, 2015-10-08:
When interacting with ROS, one key thing to know is that you have to give up on the idea of making connections yourself. ROS wants to manage them.  So a sequence like `yarpview --name /view` `yarp connect /topic@/node /view` won't lead to good outcomes. Rather you'd need to name the viewer port in a way ROS can understand, like `yarpview --name /topic@/viewer` and then let it make the needed connection.

The same issue is happening with `yarp read /x /t@/n`, it is trying to make a connection, but that won't work for ROS stuff. `yarp read /t@/n` works (and the c++ code works) because it just names the port in a ROS-compatible way, and then the connection is instigated by `roscore`.

plinioMoreno, 2015-10-09:
Many thanks @paulfitz for your explanation. I would suggest to add this explanation to the YARP with ROS wiki page as another section or in a kind of questions and answers section.

I'll run the tests on the simple example to confirm and let you know.

barbalberto, 2015-10-09:
@paulfitz  I'm not sure to understand the difference between:
```yarp read /t@/n``` and ```yarp read /x /t@/n + connect```

Is it only a matter of portname?
Will it be possible in your opinion to detect that the connection is between ROS topic and yarp port and add a "fake" name to the port for ros sake? I mean, a single port with both a yarp name and its translate ros version.
I think it'll be nice to use the connect if data are compatible and for some reason I cannot change the portname, or however it'll be more user-friendly. @lornat75 what do you think?

paulfitz, 2015-10-09:
In the `yarp read /t@/n` case, there is a port with name `/t@/n` (which ROS
will be able to understand and connect to/from).  In the `yarp read /x ...`
case there is a port with name `/x` which ROS won't be able to connect to
(what topic or service would it be? what is the node name?).  It is in fact
possible to make this work by adding modifiers into the connection type or
by subscribing to the topic separately, but it is really going against the
grain for how ROS wants the world to be.

It would be possible to add a ROS alias for port names that cannot be
changed. It'd need some work to manage the two sets of names.

lornat75, 2015-10-20:
Looks like only #589 remain to be addressed, right? Can we then close this issue?

plinioMoreno, 2015-10-20:
Yes, you can close it.

closed 2015-10-21 by lornat75