robotology/yarp #592: brew install yarp failing on Travis workers

traversaro, 2015-09-25:
See https://travis-ci.org/robotology/codyco-superbuild/jobs/82131783
~~~
==> Installing homebrew/x11/yarp
==> Cloning https://github.com/robotology/yarp.git
Cloning into '/Library/Caches/Homebrew/yarp--git'...
remote: Counting objects: 4234, done.
remote: Compressing objects: 100% (3072/3072), done.
remote: Total 4234 (delta 1232), reused 2862 (delta 1022), pack-reused 0
Receiving objects: 100% (4234/4234), 12.64 MiB | 10.90 MiB/s, done.
Resolving deltas: 100% (1232/1232), done.
Checking connectivity... done.
Checking out files: 100% (3937/3937), done.
==> Checking out branch master
==> cmake -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL_PR
Last 15 lines from /Users/travis/Library/Logs/Homebrew/yarp/01.cmake:
   ""
  is not set to a MacOSX SDK with a recognized version.  Either set
  CMAKE_OSX_SYSROOT to a valid SDK or set CMAKE_OSX_DEPLOYMENT_TARGET to
  empty.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.3.2/share/cmake/Modules/CMakeSystemSpecificInformation.cmake:36 (include)
  CMakeLists.txt:3 (project)
CMake Error: Internal CMake error, TryCompile configure of cmake failed
-- Configuring incomplete, errors occurred!
See also "/tmp/yarp20150925-12405-y4epum/CMakeFiles/CMakeOutput.log".
See also "/tmp/yarp20150925-12405-y4epum/CMakeFiles/CMakeError.log".
READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/homebrew/homebrew-x11/issues
The command "sh .ci/travis-deps.sh" failed and exited with 1 during .
~~~

traversaro, 2015-09-25:
fyi @kt10aan @francesco-romano

traversaro, 2015-09-30:
Reminder: revert https://github.com/robotology/codyco-superbuild/commit/66ffa5616ab423b4f5f8d2a20e312ba95072568e when this bug is fixed .

francesco-romano, 2015-09-30:
Able to replicate.

francesco-romano, 2015-09-30:
After a bit of debugging with @traversaro it seems that cmake fails to properly set the `CMAKE_OSX_SYSROOT` if the generator is not `Xcode`.
Probably this is a bug in cmake. But this should be investigated.

One possible solution can be to generate the bundles only if the generator is `Xcode`. But this means that the brew formula should be changed to generate and Xcode project if we want to generate the bundles.

Furthermore, I saw this note in the cmake documentation regarding both `CMAKE_OSX_SYSROOT` and `CMAKE_OSX_DEPLOYMENT_TARGET`:

~~~
The value of this variable should be set prior to the first project() or enable_language() command invocation because it may influence configuration of the toolchain and flags. It is intended to be set locally by the user creating a build tree.
~~~
But we set it after `project(YARP)`, so this can be the issue.

@drdanz @kt10aan

francesco-romano, 2015-09-30:
Last update on the test:
If the deployment target is set before the project, than everything works as expected (a part the first warning). I want to stress that this issue happens only when using `Unix Makefiles` generator and not `Xcode`.

I'll open an issue in cmake.

traversaro, 2015-09-30:
The CMake issue tracker is available at http://public.kitware.com/Bug .

kt10aan, 2015-09-30:
We could add an option to the homebrew formula for the generators.

By the way, I am testing the El Capitan right now with homebrew and the yarp formula to see if everything works ok, with the new system and the new Xcode. From past experience, the first days of a new OS X version are a bit rocky, regarding homebrew

francesco-romano, 2015-10-03:
Before firing the bug in the Cmake issue tracker, I'd like to understand a bit better the issue.
Can you please help me in this?

From what I'm understanding in [DarwinInitialize](https://github.com/Kitware/CMake/blob/master/Modules/Platform/Darwin-Initialize.cmake#L48-L51) in case we use Unix Makefiles we do not enter in that if (while we enter it if we use the Xcode generator or if we set the deployment target BEFORE calling project)
Now the question is: do we really need the sysroot variable? For sure we need the deployment target. 
Is this enough to fire a bug?

@drdanz @traversaro

traversaro, 2015-10-03:
The original commit [1] was made for 
~~~
Initialize variables CMAKE_OSX_SYSROOT, CMAKE_OSX_DEPLOYMENT_TARGET, and
CMAKE_OSX_ARCHITECTURES prior to enabling any languages.  This will
allow compiler identification to consider these values.
~~~
If this variable are not initialized then there is a problem? 

[1]  https://github.com/Kitware/CMake/commit/0200d0a9c46da512a64aed381fd2e8781e04e32e

francesco-romano, 2015-10-03:
That is what I'm trying to understand.
Do you think this is enough to open an issue in CMake?

traversaro, 2015-10-03:
I don't know, perhaps people in the CMake bugtracker can provide more insight.

francesco-romano, 2015-10-03:
ok. I'll proceed to open an issue there.

kt10aan, 2015-10-05:
What is interesting though, is that ``brew install yarp --whatever-option``(so it is compiles from source) does not fail. Only when the ``--HEAD`` option is used.

francesco-romano, 2015-10-05:
Yes, because only in head there is the CMAKE_DEPLOYMENT_TARGET option

ChiaraBartolozzi, 2015-10-08:
Ciao Guys, I had the same problem when generating for Xcode....
running cam twice solves the problem (sort of), as as soon I try to configure this is the message:

 CMake Warning at /usr/local/Cellar/cmake/3.0.2/share/cmake/Modules/Platform/Darwin.cmake:179 (message):
   Ignoring CMAKE_OSX_SYSROOT value:

    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk

   because the directory does not exist.
 Call Stack (most recent call first):
   /usr/local/Cellar/cmake/3.0.2/share/cmake/Modules/CMakeSystemSpecificInformation.cmake:36 (include)
   CMakeLists.txt:17 (project)

apparently the problem is not solved, just ignored.
I don't know if this will give problems, just adding an extra piece of information to this issue :-)

ChiaraBartolozzi, 2015-10-08:
I can't believe it, but I solved it!!!!
just put export MACOSX_DEPLOYMENT_TARGET=10.11
in the .bash_profile

francesco-romano, 2015-10-17:
Should have been fixed (for brew - travis) with https://github.com/robotology/yarp/pull/606

closed 2015-10-17 by francesco-romano

traversaro, 2015-10-17:
Strictly speaking `brew install yarp --HEAD` is still failing for Travis workers ( https://travis-ci.org/robotology/codyco-superbuild/jobs/85616012 ) but this is due to https://github.com/robotology/yarp/issues/598 , so I guess it is ok to close this issue.

francesco-romano, 2015-10-17:
The error is in iCub not in yarp.

traversaro, 2015-10-17:
Right!