robotology/yarp #594: Cannot get numpy matrix from YARP Mono16 image

jgvictores, 2015-10-01:
What I need is to treat a YARP Mono16 image (obtained from the YARP OpenNI2 device) within Python.

The full output from running an improvised yarp2py_mono16.py (https://gist.github.com/jgvictores/faafd3dbb84c0209e379) is the following:
```read() reallocated my yarp_image!
Traceback (most recent call last):
  File "yarp2py_mono16.py", line 32, in <module>
    matplotlib.pylab.imshow(img_array,interpolation='none')
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 2961, in imshow
    imlim=imlim, resample=resample, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/axes/_axes.py", line 4645, in imshow
    im.set_data(X)
  File "/usr/lib/python2.7/dist-packages/matplotlib/image.py", line 438, in set_data
    raise TypeError("Invalid dimensions for image data")
TypeError: Invalid dimensions for image data```
This leads me to believing the initial message is not the only issue in my code.

I'm using Python 2.7.9. Strangely enough, for RGB images `yarp_image.setExternal(img_array, img_array.shape[1], img_array.shape[0])` works, but the recommended for Python < 3 (`yarp_image.setExternal(img_array.__array_interface__['data'][0], img_array.shape[1], img_array.shape[0])`) leads to a segfault (example at https://gist.github.com/jgvictores/2c6481123ee56a9ea435).

Even my attempts at touching yarp.i to create a Mono16 getPixel function (and then just loop through everything, as my requirements are not time-critical) have given bogus results...

Any pointers, ideas or help please? (wasn't sure if this was Q/A; posting here since it's code-related)

plinioMoreno, 2015-10-06:
Hi @jgvictores ,

have a look to the #181, where I realized that the 2.7.9 mapped back some functions from Python 3, so the Python <3 old code does not work.

My guess is that the line

https://gist.github.com/jgvictores/faafd3dbb84c0209e379#file-yarp2py_mono16-py-L20

needs to be changed to 
``img_array = numpy.zeros((240, 320), dtype=numpy.uint16)``

Because if you add a third dimension (1), that creates the python array in a different way.

Hope this helps

jgvictores, 2015-10-08:
@plinioMoreno Thank you, that worked out so nicely!
PS: Updated gist to working version: https://gist.github.com/jgvictores/faafd3dbb84c0209e379

closed 2015-10-08 by jgvictores